node-opcua/node-opcua #673: Timeout .... waiting for response for  PublishRequest

fossejc, 2019-09-19:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting

**Current behavior**

My OPC Client gets a timeout on a client subscription error after running for a while.

```

Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module ERROR INFO: Timeout opc/:OPC.SimaticNET.TCSB error
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  authenticationToken           /* NodeId              */: ns=0;i=2776960097
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  timestamp                     /* DateTime            */: 2019-09-18T23:28:28.287Z
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  requestHandle                 /* UInt32              */: 423               0x1a7
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  returnDiagnostics             /* UInt32              */: 0               0x0
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  auditEntryId                  /* UAString            */: 
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  timeoutHint                   /* UInt32              */: 72000               0x11940
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO  additionalHeader              /* ExtensionObject     */: null
Sep 19 01:29:40 beta.thinkee.io beta-thinkee_opc-module INFO };
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  authenticationToken           /* NodeId              */: ns=0;i=2776960097
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  timestamp                     /* DateTime            */: 2019-09-18T23:28:33.727Z
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  requestHandle                 /* UInt32              */: 424               0x1a8
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  returnDiagnostics             /* UInt32              */: 0               0x0
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  auditEntryId                  /* UAString            */: 
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  timeoutHint                   /* UInt32              */: 72000               0x11940
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO  additionalHeader              /* ExtensionObject     */: null
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module ERROR INFO: Timeout opc/:OPC.SimaticNET.TCSB error
Sep 19 01:29:45 beta.thinkee.io beta-thinkee_opc-module INFO };
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  authenticationToken           /* NodeId              */: ns=0;i=2776960097
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  timestamp                     /* DateTime            */: 2019-09-18T23:28:39.166Z
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  requestHandle                 /* UInt32              */: 425               0x1a9
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  returnDiagnostics             /* UInt32              */: 0               0x0
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  auditEntryId                  /* UAString            */: 
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  timeoutHint                   /* UInt32              */: 72000               0x11940
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO  additionalHeader              /* ExtensionObject     */: null
Sep 19 01:29:51 beta.thinkee.io beta-thinkee_opc-module INFO };
```

**Expected behavior**

I expect the client to reconnect when such a timeout appear.

**Minimal reproduction of the problem with instructions**

  1. Create a client and connect it to a server then subscribe and wait. I am not able to reproduce this error demand. It may comes from a configuration error ...
 
```
   const opcua = require('node-opcua'); // eslint-disable-line
   const name = 'name';
    const client = opcua.OPCUAClient.create({
      connectionStrategy: {
        maxRetry: 1000,
        initialDelay: 1000,
        maxDelay: 30000,
      },
      keepSessionAlive: true,
      clientName: name,
      endpoint_must_exist: false,
      securityMode: opcua.MessageSecurityMode.None,
      securityPolicy: opcua.SecurityPolicy.None,
    });

    this.client.on('timed_out_request', async () => {
      console.error(`INFO: Timeout ${name} error`);
    });

    const session = await client.createSession({
      userName: 'username',
      password: 'password',
    });
    const subscription = opcua.ClientSubscription.create(session, {
      requestedPublishingInterval: 900,
      requestedLifetimeCount: 100,
      requestedMaxKeepAliveCount: 2,
      maxNotificationsPerPublish: 10,
      publishingEnabled: true,
      priority: 10,
    });
   subscription.on('started', async () => {
      console.info(`INFO: Session ${session.name} Subscribed - subscriptionId as ${this.subscription.subscriptionId} - Starting commands`);
    })
```

**Please tell us about your environment:**

  * (x ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x ) I have installed node-opcua as a package ( using npm install )

  * Device: Serveur Linux
  * OS version: Docker alpine
    - (x ) Linux   : version : Alpine
     
    
* **node-opcua version:**  :

2.1.0

* **Node:** 
   `node --version` = 12

erossignon, 2019-10-19:
can you provide a functional test script ?

fossejc, 2019-10-20:
This is the script I use, it may come for the OPC server I tried with.
I was able to reconnect to the server by catching the event `timed_out_request` and recreating the session but after a while I get some certificate errors in my logs ...

I don't know how to help more ...

erossignon, 2019-10-20:
>I don't know how to help more ...

* `this.client` is not defined and causes script to crash
* await cannot be called at top level of a nodejs script file

> I was able to reconnect to the server by catching the event timed_out_request and recreating the session but after a while I get some certificate errors in my logs ...

node-opcua should take care of recreating session in case of network error ,your example do not show how you attempt this.

axi92, 2019-10-23:
I had the same problem and my fix was the http://node-opcua.github.io/api_doc/2.0.0/classes/opcuaclientbase.html#reconnectonfailure

But I have a different Problem now.
How do I handle those messages:
```
 Timeout .... waiting for response for  PublishRequest { /*RequestHeader*/
 authenticationToken           /* NodeId              */: ns=0;i=2807837217
 timestamp                     /* DateTime            */: 2019-10-23T09:34:14.641Z
 requestHandle                 /* UInt32              */: 402               0x192
 returnDiagnostics             /* UInt32              */: 0               0x0
 auditEntryId                  /* UAString            */: 
 timeoutHint                   /* UInt32              */: 50000               0xc350
 additionalHeader              /* ExtensionObject     */: null
};
```
The event 'timed_out_request' is triggered but my log is getting spamed with those every time the request is sent.

After the opcua server is online again it is reconnecting and working finde. But every night when the opcua server is offline I get those messages.

erossignon, 2020-03-22:
@axi92 @fossejc please test now with node-opcua@2.5.1 that provides many enhancements in reconnection mechanism,
If the problem persist please arrange to contact [sterfive](https://www.sterfive.com) so we can arrange a code review of your code in private.

fossejc, 2020-03-27:
@erossignon Thanks for the update.

I will try and keep you posted.

closed 2020-10-11 by erossignon