node-opcua/node-opcua #674: cant open session after recon

pankleks, 2019-09-20:
Hello,

I have question but possibly issue with creating new session after restart of OPC server.

Scenario.
- connect to OPC server
- open session
- subscribe for changes
- restart OPC server (I'm using kepware so I restart service)
- I receive: `session_close` event on session, then `connection lost` on client then `connection restored`
- till now all looks fine, connection is automatically reconnected
- now when I try to open new session I have following error:

[image]

in node_modules\node-opcua-client\source\private\opcua_client_impl.ts file.

I'm using Windows 10, node 10.9.0, opc-ua 2.1.8, OPC server -> Kepware EX6

erossignon, 2019-09-22:
please specify the parameter of the client and session object

pankleks, 2019-09-22:
Hello, thanks for fast response.

This is session:
```
requestedPublishingInterval: 1000,
requestedLifetimeCount: 1000,
requestedMaxKeepAliveCount: 10,
maxNotificationsPerPublish: 64,
publishingEnabled: true,
priority: 10
```

and connection:

```
        this._client = OPCUAClient.create({
            applicationName: "foo",
            connectionStrategy: {
                initialDelay: 1000,
                maxRetry: -1
            },
            keepSessionAlive: true,
            securityMode: MessageSecurityMode.None,
            securityPolicy: SecurityPolicy.None,
            endpoint_must_exist: false,
            requestedSessionTimeout: 60000
        });
```

pankleks, 2019-09-22:
on more observation, it only happens when I have subscription (as explained about). If there is no active subscription during restart, session opens correct.y

pankleks, 2019-10-02:
@erossignon Hello, any update on this? Thanks!

erossignon, 2019-10-05:
Will need to see your code: please contact contact [@] sterfive.com

JustMyst, 2019-10-09:
Sample Data:
Client : https://github.com/JustMyst/OPCUA-simpleClient
Server: KEPServerEx 6

Server configuration:
1. In Project > Connectivity => create Channel => Create Device => Create Tag (those 3 are specified in client.js nodeId).
* It's best to use sinus or random function, like this
[image]
2. in OPC UA Configuration add new server endpoint => localhost network adapter, with security policy: NONE
[image]

Client Configuration:
1. Specify `nodeId` in client.js  - `nodeId="ns=2;s=<channel>.<device>.<tag>"`
2. Specify `endpointUrl` - server endpoint 

Steps:
1. Run server
2. Run Client
* Wait for client to connect and establish subscription
3. Stop Server's Runtime Service
* Client should disconnect and close session, then it tries to reconnect
4. Start Server's Runtime Service
* Client has 'connection_reestablished` event and tries to create new session and subscribe

**Error during creating new session - `failed to disconnect exiting sessions`**

JustMyst, 2019-10-11:
@wilclefison That's almost same as already done, and written in Step 4 + Error. 

Point is, that on creating new session (With this method also) after reconnecting, it throws error, that old session could not be disconnected properly.

erossignon, 2019-10-17:
In fact, it seems that KepwerServerEx may invalidate Session too early, once reconnection is made ...
A fix to handle this situation seemlessly is being implemented and test in  a separated branch... keep posted. 
NodeOPCUA user  should not need to recreate session themself and node-opcua shall take care of reconnection

pankleks, 2019-10-17:
awesome news!

pankleks, 2019-10-24:
@erossignon any update on progress?

erossignon, 2020-01-12:
should be fixed in version 2.4.1,  @pankleks  can you test it further and confirm ?

JustMyst, 2020-01-14:
**Before update:**
[image]

**After update to 2.4.1:**
[image]

**Problem is Solved** as no error is thrown and new sessions for subscriptions are created immediately after connection to server is reestablished.

@erossignon Thanks for your help with this issue!

erossignon, 2020-01-14:
you're welcome ! [:rose:](https://github.com/sponsors/node-opcua)

closed 2020-01-14 by erossignon

sukkamehu, 2020-11-06:
I'm in pretty much similar troubles. Everything works fine until I restart my simulator server or hard reboot PLC.
Once reconnecting again due to my understanding UA client should be able to recover itself but also I tried to disconnect and flush all subscriptions and sessions w/ new connection by ending up with the below results.

I've tried below subscription methods by ending with the same reults:
```
	  /* SET SUBSCRIPTION */
	  Subscription[uaEquipment] = await OPCUA.ClientSubscription.create(Session[uaEquipment], {
	    requestedPublishingInterval: Session[uaEquipment].sample_interval*1000,
	    requestedLifetimeCount: 100,
	    requestedMaxKeepAliveCount: 10,
	    minimumSamplingInterval: 10000,
	    maxNotificationsPerPublish: 100,
	    publishingEnabled: true,
	    priority: 10
	  });

	Subscription[uaEquipment] = await Session[uaEquipment].createSubscription2({
	    maxNotificationsPerPublish: 100,
	    publishingEnabled: true,
	    requestedLifetimeCount: 100,
	    requestedPublishingInterval: Session[uaEquipment].sample_interval*1000,
	    requestedMaxKeepAliveCount: 10,
	});
```

In below is the thing. Before restart I get clear OPC UA messages with the variable data but afterwards just event notifications.

```
0|OPC UA  |       { /*4*/
0|OPC UA  |         clientHandle              /* UInt32              */: 5               0x5
0|OPC UA  |         value                     /* DataValue           */: { /* DataValue */
0|OPC UA  |       value: Variant(Scalar<Float>, value: 0.39416423439979553)
0|OPC UA  |       statusCode:      Good (0x00000)
0|OPC UA  |       serverTimestamp: 2020-11-06T14:04:11.903Z $ 215.299.000
0|OPC UA  |       sourceTimestamp: 2020-11-06T14:04:11.819Z $ 843.788.000
0|OPC UA  |    }
0|OPC UA  |       }
0|OPC UA  |     ]
0|OPC UA  |     diagnosticInfos               /* DiagnosticInfo   [] */: [ /* empty*/ ]
0|OPC UA  |    };
0|OPC UA  |  ]
0|OPC UA  | };
0|OPC UA  | 2020-11-06T14:04:12.893Z Connection lost to opc.tcp://192.168.8.113:4840 (Simulation)
0|OPC UA  | 2020-11-06T14:04:12.895Z Trying to connect opc.tcp://192.168.8.113:4840 (Simulation)
0|OPC UA  | 2020-11-06T14:04:12.897Z Attempt 0 connecting to opc.tcp://192.168.8.113:4840, next attempt in 1 seconds
0|OPC UA  | 2020-11-06T14:04:13.899Z Attempt 1 connecting to opc.tcp://192.168.8.113:4840, next attempt in 2 seconds
0|OPC UA  | 2020-11-06T14:04:15.933Z Connection re-established to opc.tcp://192.168.8.113:4840 (Simulation)
0|OPC UA  | { /*NotificationMessage*/
0|OPC UA  |  sequenceNumber                /* UInt32              */: 1               0x1
0|OPC UA  |  publishTime                   /* DateTime            */: 1601-01-01T00:00:00.000Z
0|OPC UA  |  notificationData              /* ExtensionObject  [] */: [ /* empty*/ ]
0|OPC UA  | };
0|OPC UA  | { /*NotificationMessage*/
0|OPC UA  |  sequenceNumber                /* UInt32              */: 1               0x1
0|OPC UA  |  publishTime                   /* DateTime            */: 1601-01-01T00:00:00.000Z
0|OPC UA  |  notificationData              /* ExtensionObject  [] */: [ /* empty*/ ]
0|OPC UA  | };
```

Can this be caused possibly due to my method of storing different sessions and subscriptions on objects as otherwise I think I'm following tutorials and examples?

In below part of the code where I guess the issue is existing most likely
```
	  /* LIST ITEMS FOR BEING MONITORED */
	  let itemsToMonitor = [];
	  
	  target.tags.forEach(function(node, index){
		itemsToMonitor.push({
		  dbId: node.dbId,
		  name: node.name,
		  type: node.type,
		  description: node.description,
		  events: node.events,
		  nodeId: node.nodeId,
		  attributeId: OPCUA.AttributeIds.Value
		});
	  });

	  console.log(itemsToMonitor)
	  
	  /* OPTIONS FOR MONITOR GROUP */
	  const optionsGroup = {
        discardOldest: true,
        queueSize: 1,
		samplingInterval: target.sample_interval*1000,
      };
	  
	  /* CREATE ITEM GROUP FOR MONITORED ITEMS */
	  let monitoredItemGroup = await OPCUA.ClientMonitoredItemGroup.create(Subscription[uaEquipment], itemsToMonitor, optionsGroup, OPCUA.TimestampsToReturn.Both);
	  

		monitoredItemGroup.on("initialized", () => {
			console.log(new Date(),`Started monitoring values on ${target.host} (${uaEquipment}), ${Session[uaEquipment]}`);
		});

        monitoredItemGroup.on("changed", (monitoredItem, dataValue, index) => {
			if(dataValue.sourcePicoseconds === 0){
			  dataValue.sourcePicoseconds = (new Date()).getTime();
		    };
				//console.log(dataValue)
			dataParser(dataValue, itemsToMonitor[index], target.storedb);
		    		
        });
```