node-opcua/node-opcua #677: [Suggestion] Creating _enumerationSecurityPolicy

gabrielmbmb, 2019-09-29:
Hello!

First of all, great work! The library is awesome for working with OPC UA.

I was creating a client and I used the _enumerationMessageSecurityMode to show the possible options to the user in case an invalid option is passed to coerceMessageSecurityMode. I wanted to do the same but with coerceSecurityPolicy, but it seems there is no _enumerationSecurityPolicy. At this moment I'm iterating over the object SecurityPolicy and printing its keys.

Thank you so much!

gabrielmbmb, 2019-09-29:
Also, when is a SecurityPolicy considered Invalid? I've seen [coerceSecurityPolicy function declaration](https://github.com/node-opcua/node-opcua/blob/eeade1b9/packages/node-opcua-secure-channel/source/security_policy.ts#L144) and it will never return SecurityPolicy.Invalid.

erossignon, 2019-10-05:
_enumerationMessageSecurityMode  starts with an underscore, this means it is not part of the public interface and subject to change. I would suggest not to use it.

You can use this for instance: 
```` 
> const opcua = require("node-opcua");
> Object.keys(opcua.SecurityPolicy)   
[ 'Invalid',                          
  'None',                             
  'Basic128',                         
  'Basic192',                         
  'Basic192Rsa15',                    
  'Basic256Rsa15',                    
  'Basic256Sha256',                   
  'Aes128_Sha256_RsaOaep',            
  'PubSub_Aes128_CTR',                
  'PubSub_Aes256_CTR',                
  'Basic128Rsa15',                    
  'Basic256' ]                        
>                                     
```

gabrielmbmb, 2019-10-05:
Perfect! Thanks you!

closed 2019-10-05 by gabrielmbmb