node-opcua/node-opcua #678: Client reconnection doesn't recreate subscriptions

netbymatt, 2019-09-30:
If I intentionally shut down my OPC-UA server (Siemens S7-1500) and restart it, the code noted below triggers.
https://github.com/node-opcua/node-opcua/blob/9d092fd1fe78a78cd87fad584b5386e3d9bd5036/packages/node-opcua-client/source/reconnection.ts#L215-L219
The server responds with BadServiceUnsupported (0x800b0000) in response to TransferSubscriptionsRequest which then causes the callback to be triggered and the error passed upstream. Based on the comment within this block and the subsequent code, the BadServiceUnsupported response should not trigger an error response and should allow the remaining code in the block to attempt to recreate the subscriptions.

Am I misunderstanding how this supposed to work?

erossignon, 2019-10-01:
Good point, it seems that Siemens does not implement the TransferService .. 
Will need to fix it.

erossignon, 2019-10-02:
Hi @netbymatt  , fix is in v2.1.9 , can you help us and do a little bit of testing to verify that is is working now ?

netbymatt, 2019-10-02:
Thanks for your quick response! I'll give it a try in the next few days when I'm at a good point to take a processor offline for a little while.

netbymatt, 2019-10-03:
This fix is confirmed. I tested three scenarios: Pulling the ethernet cable from the PLC, downloading "software changes only" to the PLC (which I observed causes the OPCUA server to restart), and power cycling the PLC.

For what it's worth I also observed that the OPCUA server in the PLC can be connected to up to a minute before Siemens own programming software can connect to the PLC. It appears (although I didn't test if fully) that the OPCUA server can read tags but they are frozen at their default states as the PLC has not started executing code yet. Nothing to fix with this, just an observation for anyone who may come across this thread while looking into similar reconnection issues.

closed 2019-10-03 by netbymatt

teropes, 2019-10-07:
@netbymatt Hi, we are also using Siemens S7 PLC's with node-opcua applications and have struggled with similar issues. It's great that you have spotted this and also verified the fix for it.

I would be also interested to know that have you used node-opcua for reading / writing any custom data types of Siemens S7 PLC? I'm still using v0.7.2 with our custom opcua.types.js hack to be able to read / write custom datatypes, but its breaking in any of the more recent versions and its preventing me from upgrading to newer versions.

netbymatt, 2019-10-07:
@teropes Unfortunately I'm not reading the custom data types directly. All of the tags I'm reading are part of custom data types, but I only read the elementary data types within them.