node-opcua/node-opcua #679: How to use the crawler?

binaryPUNCH, 2019-10-01:
https://github.com/node-opcua/node-opcua/blob/9d092fd1fe78a78cd87fad584b5386e3d9bd5036/packages/node-opcua-client-crawler/source/node_crawler.ts#L306-L312

I'm not quite sure how to use the crawler. It want's the type NodeCrawlerClientSession to instantiate NodeCrawler, however I only know how to create a ClientSession

binaryPUNCH, 2019-10-01:
`  var crawler = new NodeCrawler(<NodeCrawlerClientSession><unknown>session);` does not seem to work @erossignon

erossignon, 2019-10-02:
how about
```typescript
import { OPCUAClient, NodeCrawler } from "node-opcua";
import * as path from "path";

(async () => {
    try {
        const client = OPCUAClient.create({
            endpoint_must_exist: false
        });

        const endpoint = "opc.tcp://localhost:4334";
        await client.connect(endpoint);

        const session = await client.createSession();

        const crawler = new NodeCrawler(session);

        const data = await crawler.read("ns=0;i=80");
        console.log(data);

        await session.close();
        await client.disconnect();
    } catch (err) {
        console.log("err = ", err.message);
    }
})();
```
?

binaryPUNCH, 2019-10-02:
Thanks for the fast response. Really amazing how much you've done for this project. Unfortunately I get `Property 'create' does not exist on type 'typeof OPCUAClient'.ts(2339)` 

Am I using too new a version     

"node-opcua-client": "^2.1.8",
"node-opcua-client-crawler": "^2.1.8",

erossignon, 2019-10-02:
Please use the factory constructor, it should be   
 ```const client = OPCUAClient.create({
            endpoint_must_exist: false
        });
```

binaryPUNCH, 2019-10-02:
That doesn't work. It appears as if it exists in the latest version:

`Property 'create' does not exist on type 'typeof OPCUAClient'.ts(2339)`

erossignon, 2019-10-02:
you can make it work with node-opcua-client by replacing
the first line 
```
import { OPCUAClient, NodeCrawler } from "node-opcua";
```

```
import { OPCUAClient } from "node-opcua-client";
import { NodeCrawler } from "node-opcua-client-crawler";
```
Strange , the script provided provided works on myside. There must be something else ? 
Can you provide a complete example that reproduce the issue  (including package,json, tsconfig,json ts files etc...) ?

binaryPUNCH, 2019-10-04:
So weird. Something must have been wrong with my VS Code editor. Now I can use the factory create function and use the session type matches the NodeCrawler constructor.

binaryPUNCH, 2019-10-04:
Is there any way to throttle the crawler, to not bring down the whole OPCUA server if there are hundreds of thousands of tags?

binaryPUNCH, 2019-10-07:
Also, is there any way to get the "AttributeIds.Description" out? It doesn't seem like crawler.read takes the ReadValueIdLike, unlike the .read() function directly on the session.

erossignon, 2019-10-09:
> Is there any way to throttle the crawler, to not bring down the whole OPCUA server if there are hundreds of thousands of tags?
Yes , use 
````
 const data = {
            onBrowse(this: UserData, crawler1: NodeCrawler, cacheNode: CacheNode) {
                results.push(cacheNode.browseName.toString());
                if(someconditiion) {
                   followForward(crawler1, cacheNode, this);
                }
            }
        };

        await crawler.crawl(nodeId, data);
````
> Also, is there any way to get the "AttributeIds.Description" out? It doesn't seem like crawler.read takes the ReadValueIdLike, unlike the .read() function directly on the session.
No I am aware of , you can always query for description once you have the tree for the nodeId you are interested of

closed 2019-10-09 by binaryPUNCH

binaryPUNCH, 2019-12-19:
If anyone lands here via. Google, small change I made to get the displayName and description out: https://github.com/binaryPUNCH/node-opcua/commit/d7f815cb41cca368c764eea40baa9dbf044f6b02

schaefersimon, 2020-06-23:
It appears that the instantiation of the NodeCrawler class in node-opcua@2.6.1 changed since the constructor expects a NodeCrawlerClientSession instead of the ClientSession.

How do I have to change the following code to use the NodeCrawler in version 2.6.1:

`import { OPCUAClient, NodeCrawler } from "node-opcua";
import * as path from "path";

(async () => {
    try {
        const client = OPCUAClient.create({
            endpoint_must_exist: false
        });

        const endpoint = "opc.tcp://localhost:4334";
        await client.connect(endpoint);

        const session = await client.createSession();

        const crawler = new NodeCrawler(session);

        const data = await crawler.read("ns=0;i=80");
        console.log(data);

        await session.close();
        await client.disconnect();
    } catch (err) {
        console.log("err = ", err.message);
    }
})();`