node-opcua/node-opcua #680: Issue with request timeout

Odenelle, 2019-10-01:
**I'm submitting a feature request

**Current behavior**

We are facing an unusual behavior of our OPC-UA server (which is managed by another team): it exposes the data with an interval of 10 minuts.
Between each interval, the OPCUA server doesn’t store the data filled by the request, and the OPCUA READ requests are falling in timeout.

This OPC UA server requests the data from an external database, which is overloaded. 

For this reason, we cannot reduce the interval of requests from the kepserver to the database: it is 10 minuts.

We realized that the interval between each data availability (that never last more than one second) correspond to the interval of SQL request: for this reason we cannot change it.

We were forced to hardcode the defaultTransactionTimeout (set to 60 by default) inside node-opcua-secure-channel/src/client/client_secure_channel_layer.js to 60000 in order to make sure that the read request are not permanently falling in timeout.

**Expected behavior**

We would like to have the capability to change the transaction timeout without modifying the node-opcua library.

**Minimal reproduction of the problem with instructions**
We cannot give steps to reproduce the issue: the behavior our OPCUA server is the cause.
If you try to read nodes from an OPCUA Server that serves the data only every 10 minutes, then you should face the same issue: timeouts for each read requests that did not occur 1 minute before the data serving.
  
**What is the motivation / use case for changing the behavior?**
We are in an industrial context, this database is overloaded and our OPCUA client use case is related to this SQL request that is made from our OPCUA Server to the database.
We think it would be great to let the possibility to change the transactionTimeout from our code to avoid this hardcoded workaround.

**Please tell us about your environment:**
Ubuntu 18.04 LTS
Kepserver
Node package manager

  * ( ) my request is related to node-opcua acting as an  OPCUA CLIENT 
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
        
    - ( ) Linux   : version : Ubuntu 18.04 LTS

   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Kepserver
     - Version: 6.7.1046.0
     - Manufacturer/Software vendor: PTC
     - link : https://www.kepware.com/en-us/kepserverex-6-7-release/
     
    
* **node-opcua version:**  :
"node-opcua": "^0.5.5"

* **Node:** 
   `node --version` = v10.6.0

erossignon, 2019-10-02:
Server should be able to return BadDataIsNotAvaiable rather than blocking the transaction for up to 10 minutes waiting for the data to be availabe. There must be a way to intervene on the design of the server.

erossignon, 2020-03-20:
Alternatively, a monitored item could be used to subscript to the data. In this scenario the database reading is initiated asynchronously outside any opcua read request and in this case the client will be automatically updated when the result is ready.

closed 2021-01-17 by erossignon