node-opcua/node-opcua #681: Authentication

ParkerOu, 2019-10-03:
**I'm submitting a ...** 
- (x) bug report

**Current behavior**
run sample client

**Expected behavior**
subscribe the server and get tag value

**Minimal reproduction of the problem with instructions**

```js
const opcua = require('node-opcua');

const endpointUrl = 'opc.tcp://localhost:49320';
const client = opcua.OPCUAClient.create({
    securityMode: opcua.MessageSecurityMode.None,
    securityPolicy: opcua.SecurityPolicy.None,
    endpoint_must_exist: false
});
```
I created client but received below error:
```
cannot locate certificate file D:\work\test\node-opcua-v2.x\node_modules\node-opcua-client\certificates\client_selfsigned_cert_2048.pem
```
related code:

> https://github.com/node-opcua/node-opcua/blob/v2.1.9/packages/node-opcua-client/source/private/client_base_impl.ts#L307-L315
> 

**Please tell us about your environment:**
  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x) I have installed node-opcua as a package ( using npm install )
  * OS version: _____
    - (x) Windows : version : 10
   
   * Description of the other OPCUA system I am trying to connect to:
Kepware Server 
     
    
* **node-opcua version:**  :
v2.1.9

* **Node:** 
v10.11.0

ParkerOu, 2019-10-04:
In this case, authentication will be Anonymous.
No need for certificate.
I suppose node-opcua can use one of this authentication
1. Anonymous
2. User Name Password
3. X509 Certificate

erossignon, 2019-10-05:
Certificate is always needed .

Make sure you pass valid certificate file and private key file while creating your client.

If you don't default self-signed certificate created at installation time will be used.
For some reason, those certificates do not exist on your machine. I see you are running on Win10. 
Are you behind a proxy ?

erossignon, 2019-10-19:
@ParkerOu  Are you behind a proxy ?

ParkerOu, 2019-10-19:
@erossignon  Sorry for being slow to reply.
I used in locally.
OS: Win 10 Pro

ParkerOu, 2019-11-19:
The problem was solved by `npm i` again.

ParkerOu, 2020-01-07:
- node-opcua version: v2.4.1

Got `cannot locate certificate file D:\work\test\node-opcua-v2.x\node_modules\node-opcua-client\certificates\client_selfsigned_cert_2048.pem` again.

`npm i` throw below error:
```
node-pre-gyp WARN Using request for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(404): https://github.com/kelektiv/node.bcrypt.js/releases/download/v3.0.7/bcrypt_lib-v3.0.7-node-v64-win32-x64-unknown.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for bcrypt@3.0.7 and node@10.11.0 (node-v64 ABI, unknown) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: Command failed: C:\Users\User\AppData\Local\Programs\Python\Python37-32\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                                ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack 
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:289:12)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "D:\\useful\\node\\node.exe" "D:\\useful\\node\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=D:\\work\\API\\test\\node_modules\\bcrypt\\lib\\binding\\bcrypt_lib.node" "--module_name=bcrypt_lib" "--module_path=D:\\work\\API\\test\\node_modules\\bcrypt\\lib\\binding" "--napi_version=3" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v64"
gyp ERR! cwd D:\work\API\test\node_modules\bcrypt
gyp ERR! node -v v10.11.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'D:\useful\node\node.exe D:\useful\node\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=D:\work\API\my-dir\node_modules\bcrypt\lib\binding\bcrypt_lib.node --module_name=bcrypt_lib --module_path=D:\work\API\test\node_modules\bcrypt\lib\binding --napi_version=3 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\work\API\test\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
node-pre-gyp ERR! System Windows_NT 10.0.18362
node-pre-gyp ERR! command "D:\\useful\\node\\node.exe" "D:\\work\\API\\test\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" 
"install" "--fallback-to-build"
node-pre-gyp ERR! cwd D:\work\API\test\node_modules\bcrypt
Failed to execute 'D:\useful\node\node.exe D:\useful\node\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js configure --fallback-tonode-v64' (1)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bcrypt@3.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bcrypt@3.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\User\AppData\Roaming\npm-cache\_logs\2020-01-07T10_48_21_825Z-debug.log
```
bcrypt version 3.0.7 currently not support Windows: https://github.com/kelektiv/node.bcrypt.js/issues/766

Self-signed certificate are not created at installation, because bcrypt is not installed correctly in Windows.

ParkerOu, 2020-01-07:
@erossignon

erossignon, 2020-01-07:
@ParkerOu  I am not sure the problem is related , can you submit your package.json file for investigation ?

ParkerOu, 2020-01-08:
Here's my package.json
```json
{
  "name": "test",
  "version": "3.11.0",
  "updateDate": "2020-01-02",
  "private": true,
  "scripts": {
    "babel-node": "babel-node --presets=latest",
    "start": "babel-node src/main.js",
    "opc": "babel-node src/job/opc/subscribe.js",
    "build": "babel src --out-dir dist",
    "clean": "rm -r dist",
    "precommit": "lint-staged"
  },
  "dependencies": {
    "@sentry/node": "^5.9.0",
    "async": "^2.6.2",
    "bcrypt": "^3.0.6",
    "body-parser": "^1.18.3",
    "cookie-parser": "~1.4.3",
    "cron": "^1.7.2",
    "debug": "~2.6.0",
    "dotenv": "^8.2.0",
    "ejs": "^2.6.1",
    "express": "^4.16.3",
    "express-session": "^1.15.3",
    "fast-csv": "^3.7.0",
    "helmet": "^3.21.1",
    "influx": "^5.5.1",
    "moment": "^2.22.2",
    "morgan": "^1.9.1",
    "mqtt": "^2.18.8",
    "mysql": "^2.17.1",
    "node-opcua": "^2.4.1",
    "redis": "^2.8.0",
    "winston": "^3.1.0"
  },
  "devDependencies": {
    "@babel/cli": "^7.1.2",
    "@babel/core": "^7.1.2",
    "@babel/node": "^7.0.0",
    "@babel/polyfill": "^7.0.0",
    "@babel/preset-env": "^7.1.0",
    "@babel/register": "^7.0.0",
    "babel-eslint": "^10.0.1",
    "chai": "^4.0.2",
    "chai-http": "^4.0.0",
    "eslint": "^4.14.0",
    "eslint-config-google": "^0.9.1",
    "eslint-plugin-import": "^2.8.0",
    "eslint-plugin-sql": "^1.1.1",
    "husky": "^0.14.3",
    "lint-staged": "^7.2.2",
    "request": "^2.88.0"
  },
  "lint-staged": {
    "*.js": [
      "eslint --fix",
      "git add"
    ]
  }
}

```

ParkerOu, 2020-01-08:
Sorry for accidentally closed the issue.

erossignon, 2020-01-08:
Looking at the log the issue  comes from package "bcrypt"

``` 
npm ERR! bcrypt@3.0.7 install: `node-pre-gyp install --fallback-to-build`
``` 

Looking inside package.json, I see that you have  installed `bcrypt` .(this package is not needed by node-opcua) . I suggest that you post the issue to the bcrypt package maintainer if it is a package that you really need to use.

closed 2020-01-08 by erossignon