node-opcua/node-opcua #682: 'Server with method' doesn't return name of output arguments

christoph19, 2019-10-05:
I am trying to write an node-opcua client which invokes the bark method from "server_with_method.js".

The method is already working and returning the value of the output argument ('Whaff'). 
But I am wondering if it isn't possible to also get directly the name of the output argument (in this case 'Barks')?

Client Code:
```
const opcua = require("node-opcua");
const async = require("async");
const util = require('util')
var should = require("should");
const endpointUrl = "opc.tcp://" + require("os").hostname() + ":4334";
const client = opcua.OPCUAClient.create({
    endpoint_must_exist: false
});
client.on("backoff", (retry, delay) => 
    console.log("still trying to connect to ", endpointUrl ,": retry =", retry, "next attempt in ", delay/1000, "seconds" )
);

let the_session, the_subscription;

async.series([

    // step 1 : connect to
    function(callback)  {
        client.connect(endpointUrl, function (err) {
            if(err) {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            } else {
                console.log("connected !");
            }
            callback(err);
        });
    },

    // step 2 : createSession
    function(callback) {
        client.createSession( function(err, session) {
            if(err) {
                return callback(err);
            }
            the_session = session;
			// console.log(session);
            callback();
        });
    },
	
	
	function (callback){
		let methodToCalls = [{
			objectId: "ns=1;i=1000",
			methodId: "ns=1;i=1001",
			inputArguments: [
            //{dataType: opcua.DataType.String, value: "I"},
            {dataType: opcua.DataType.UInt32, value: 1},
			{dataType: opcua.DataType.UInt32, value: 1}]
		}];
		
		the_session.call(methodToCalls, function (err, results) {
            console.log(results.toString());
			//console.log(util.inspect(results, {showHidden: false, depth: null}))
			callback(err, results);
		});
	},
	
    // close session
    function(callback) {
        the_session.close( function(err) {
            if(err) {
                console.log("closing session failed ?");
            }
            callback();
        });
    }

],
function(err) {
    if (err) {
        console.log(" failure ",err);
    } else {
        console.log("done!");
    }
    client.disconnect(function(){});
}) ;
```
Output:

```
Variant {
  dataType: 12,
  arrayType: 1,
  value: [ 'Whaff' ],
  dimensions: null }
[ CallMethodResult {
    statusCode:
     ConstantStatusCode { _value: 0, _description: 'No Error', _name: 'Good' },
    inputArgumentResults:
     [ ConstantStatusCode { _value: 0, _description: 'No Error', _name: 'Good' },
       ConstantStatusCode { _value: 0, _description: 'No Error', _name: 'Good' } ],
    inputArgumentDiagnosticInfos: [],
    outputArguments:
     [ Variant {
         dataType: 12,
         arrayType: 1,
         value: [ 'Whaff' ],
         dimensions: null } ] } ]

```

erossignon, 2019-10-05:
what's about 
```javascript
const { OPCUAClient, AttributeIds, makeBrowsePath, StatusCodes } = require("node-opcua");

const endpointUrl = "opc.tcp://" + require("os").hostname() + ":48010";
const nodeId = "ns=2;s=Demo.Method.EnumTest";

/**
 * @param {NodeId} nodeId: nodeId of the method for which to extract arguments
 * @param {ClientSession} session: the client session
 */
async function extractMethodArguments(nodeId, session) {
    try {   
        const [input,output] =await session.translateBrowsePath([
            makeBrowsePath(nodeId,".InputArguments"), 
            makeBrowsePath(nodeId,".OutputArguments"), 
        ]);
 
        const inputArgumentNodeId  =  input.targets[0].targetId;
        const outputArgumentNodeId  =  output.targets[0].targetId;

        const nodesToRead = [
            { attributeIds: AttributeIds.Value, nodeId: inputArgumentNodeId },
            { attributeIds: AttributeIds.Value, nodeId: outputArgumentNodeId },
        ]

        const [inputArgumentValue, outputArgumentValue ]  = await session.read(nodesToRead);

        console.log("Input arguments");

        if (inputArgumentValue.statusCode === StatusCodes.Good) {
            console.log(" -> ",inputArgumentValue.value.value.map(v=>v.name).join(","));
        }

        console.log("Output arguments");
        if (outputArgumentValue.statusCode === StatusCodes.Good) {
            console.log(" <- ", outputArgumentValue.value.value.map(v=>v.name).join(","));
        }

    } catch(err) {
        console.log(err);
    }
}

(async ()=> {

    const client = OPCUAClient.create({
        endpoint_must_exist: false
    });
    client.on("backoff", (retry, delay) => 
        console.log("still trying to connect to ", endpointUrl ,": retry =", retry, "next attempt in ", delay/1000, "seconds" )
    );

    try {   
        await client.connect(endpointUrl);
        console.log("connected !");
        const session = await client.createSession();
        console.log("session created");

        await extractMethodArguments(nodeId, session);

        await session.close();
        await client.disconnect();

    } catch(err) {
        console.log(err);
    }
    
})();
```` 
( I have converted your code into modern async/await style)

christoph19, 2019-10-05:
With the extractMethodArguments function I get the attributes of the input and output arguments.
This function works pretty well and returns the name of the arguments.

But I want to get something like this after I called a method on the server:

```
outputArguments:
     [ Variant {
	//name should be added
	 name: 'Barks',
         dataType: 12,
         arrayType: 1,
         value: [ 'Whaff' ],
         dimensions: null } ]
```

Is this possible or do I need to match the attributes of the extractMethodArguments and the return values of the method?

erossignon, 2019-10-05:
Yes, you have 2 arrays, one containing the name of the arguments, the other containing the value of the arguments, just make a match with an index.

christoph19, 2019-10-06:
Ok great. This works fine, thanks for your help!

closed 2019-10-06 by christoph19