node-opcua/node-opcua #683: datatype_factory.ts error when creating server

netbymatt, 2019-10-09:
Since moving from 2.1.7 to 2.1.9 I've encountered an error when creating a server. I confirmed this by rolling back the version and running the same code to create a server.

https://github.com/node-opcua/node-opcua/blob/d79a2ddba8dc1a0b6c127bf7297d3f428db4ec41/packages/node-opcua-factory/source/datatype_factory.ts#L40-L44

Produces registerFactory: EUInformation already registered.

I'm not sure what the best information is to help debug this, but let's start here. An output of this._structureTypeConstructorByNameMap at the time the error is logged.
```
this._structureTypeConstructorByNameMap
Object {EUInformation: , Range: , AxisInformation: , FilterOperand: , SimpleAttributeOperand: , …}
[[StableObjectId]]: 8
ActivateSessionRequest: class ActivateSessionRequest extends node_opcua_extension_object_1.ExtensionObject { … }
ActivateSessionResponse: class ActivateSessionResponse extends node_opcua_extension_object_1.ExtensionObject { … }
AdditionalParametersType: class AdditionalParametersType extends node_opcua_extension_object_1.ExtensionObject { … }
AddNodesItem: class AddNodesItem extends node_opcua_extension_object_1.ExtensionObject { … }
AddNodesRequest: class AddNodesRequest extends node_opcua_extension_object_1.ExtensionObject { … }
AddNodesResponse: class AddNodesResponse extends node_opcua_extension_object_1.ExtensionObject { … }
AddNodesResult: class AddNodesResult extends node_opcua_extension_object_1.ExtensionObject { … }
AddReferencesItem: class AddReferencesItem extends node_opcua_extension_object_1.ExtensionObject { … }
AddReferencesRequest: class AddReferencesRequest extends node_opcua_extension_object_1.ExtensionObject { … }
AddReferencesResponse: class AddReferencesResponse extends node_opcua_extension_object_1.ExtensionObject { … }
AggregateConfiguration: class AggregateConfiguration extends node_opcua_extension_object_1.ExtensionObject { … }
AggregateFilter: class AggregateFilter extends MonitoringFilter { … }
AggregateFilterResult: class AggregateFilterResult extends MonitoringFilterResult { … }
Annotation: class Annotation extends node_opcua_extension_object_1.ExtensionObject { … }
AnonymousIdentityToken: class AnonymousIdentityToken extends UserIdentityToken { … }
ApplicationDescription: class ApplicationDescription extends node_opcua_extension_object_1.ExtensionObject { … }
Argument: class Argument extends node_opcua_extension_object_1.ExtensionObject { … }
AttributeOperand: class AttributeOperand extends FilterOperand { … }
AxisInformation: class AxisInformation extends node_opcua_extension_object_1.ExtensionObject { … }
BrokerConnectionTransportDataType: class BrokerConnectionTransportDataType extends ConnectionTransportDataType { … }
BrokerDataSetReaderTransportDataType: class BrokerDataSetReaderTransportDataType extends DataSetReaderTransportDataType { … }
BrokerDataSetWriterTransportDataType: class BrokerDataSetWriterTransportDataType extends DataSetWriterTransportDataType { … }
BrokerWriterGroupTransportDataType: class BrokerWriterGroupTransportDataType extends WriterGroupTransportDataType { … }
BrowseDescription: class BrowseDescription extends node_opcua_extension_object_1.ExtensionObject { … }
BrowseNextRequest: class BrowseNextRequest extends node_opcua_extension_object_1.ExtensionObject { … }
BrowseNextResponse: class BrowseNextResponse extends node_opcua_extension_object_1.ExtensionObject { … }
BrowsePath: class BrowsePath extends node_opcua_extension_object_1.ExtensionObject { … }
BrowsePathResult: class BrowsePathResult extends node_opcua_extension_object_1.ExtensionObject { … }
BrowsePathTarget: class BrowsePathTarget extends node_opcua_extension_object_1.ExtensionObject { … }
BrowseRequest: class BrowseRequest extends node_opcua_extension_object_1.ExtensionObject { … }
BrowseResponse: class BrowseResponse extends node_opcua_extension_object_1.ExtensionObject { … }
BrowseResult: class BrowseResult extends node_opcua_extension_object_1.ExtensionObject { … }
BuildInfo: class BuildInfo extends node_opcua_extension_object_1.ExtensionObject { … }
CallMethodRequest: class CallMethodRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CallMethodResult: class CallMethodResult extends node_opcua_extension_object_1.ExtensionObject { … }
CallRequest: class CallRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CallResponse: class CallResponse extends node_opcua_extension_object_1.ExtensionObject { … }
CancelRequest: class CancelRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CancelResponse: class CancelResponse extends node_opcua_extension_object_1.ExtensionObject { … }
ChannelSecurityToken: class ChannelSecurityToken extends node_opcua_extension_object_1.ExtensionObject { … }
CloseSecureChannelRequest: class CloseSecureChannelRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CloseSecureChannelResponse: class CloseSecureChannelResponse extends node_opcua_extension_object_1.ExtensionObject { … }
CloseSessionRequest: class CloseSessionRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CloseSessionResponse: class CloseSessionResponse extends node_opcua_extension_object_1.ExtensionObject { … }
ComplexNumberType: class ComplexNumberType extends node_opcua_extension_object_1.ExtensionObject { … }
ConfigurationVersionDataType: class ConfigurationVersionDataType extends node_opcua_extension_object_1.ExtensionObject { … }
ConnectionTransportDataType: class ConnectionTransportDataType extends node_opcua_extension_object_1.ExtensionObject { … }
ContentFilter: class ContentFilter extends node_opcua_extension_object_1.ExtensionObject { … }
ContentFilterElement: class ContentFilterElement extends node_opcua_extension_object_1.ExtensionObject { … }
ContentFilterElementResult: class ContentFilterElementResult extends node_opcua_extension_object_1.ExtensionObject { … }
ContentFilterResult: class ContentFilterResult extends node_opcua_extension_object_1.ExtensionObject { … }
CreateMonitoredItemsRequest: class CreateMonitoredItemsRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CreateMonitoredItemsResponse: class CreateMonitoredItemsResponse extends node_opcua_extension_object_1.ExtensionObject { … }
CreateSessionRequest: class CreateSessionRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CreateSessionResponse: class CreateSessionResponse extends node_opcua_extension_object_1.ExtensionObject { … }
CreateSubscriptionRequest: class CreateSubscriptionRequest extends node_opcua_extension_object_1.ExtensionObject { … }
CreateSubscriptionResponse: class CreateSubscriptionResponse extends node_opcua_extension_object_1.ExtensionObject { … }
DataChangeFilter: class DataChangeFilter extends MonitoringFilter { … }
DataChangeNotification: class DataChangeNotification extends NotificationData { … }
[... 203 lines cut ...]
UnregisterNodesResponse: class UnregisterNodesResponse extends node_opcua_extension_object_1.ExtensionObject { … }
UpdateDataDetails: class UpdateDataDetails extends HistoryUpdateDetails { … }
UpdateEventDetails: class UpdateEventDetails extends HistoryUpdateDetails { … }
UpdateStructureDataDetails: class UpdateStructureDataDetails extends HistoryUpdateDetails { … }
UserIdentityToken: class UserIdentityToken extends node_opcua_extension_object_1.ExtensionObject { … }
UserNameIdentityToken: class UserNameIdentityToken extends UserIdentityToken { … }
UserTokenPolicy: class UserTokenPolicy extends node_opcua_extension_object_1.ExtensionObject { … }
VariableAttributes: class VariableAttributes extends NodeAttributes { … }
VariableNode: class VariableNode extends InstanceNode { … }
VariableTypeAttributes: class VariableTypeAttributes extends NodeAttributes { … }
VariableTypeNode: class VariableTypeNode extends TypeNode { … }
ViewAttributes: class ViewAttributes extends NodeAttributes { … }
ViewDescription: class ViewDescription extends node_opcua_extension_object_1.ExtensionObject { … }
ViewNode: class ViewNode extends InstanceNode { … }
WriteRequest: class WriteRequest extends node_opcua_extension_object_1.ExtensionObject { … }
WriteResponse: class WriteResponse extends node_opcua_extension_object_1.ExtensionObject { … }
WriterGroupDataType: class WriterGroupDataType extends PubSubGroupDataType { … }
WriterGroupMessageDataType: class WriterGroupMessageDataType extends node_opcua_extension_object_1.ExtensionObject { … }
WriterGroupTransportDataType: class WriterGroupTransportDataType extends node_opcua_extension_object_1.ExtensionObject { … }
WriteValue: class WriteValue extends node_opcua_extension_object_1.ExtensionObject { … }
X509IdentityToken: class X509IdentityToken extends UserIdentityToken { … }
XVType: class XVType extends node_opcua_extension_object_1.ExtensionObject { … }
__proto__: Object {constructor: , __defineGetter__: , __defineSetter__: , …}

```

erossignon, 2019-10-09:
Please check that all node-opcua modules in your package.json file have the latest version.
Use npm-check-updates for instance to migrate them all safely.

netbymatt, 2019-10-10:
This is resolved. I'm not sure how I got things out of sync.