node-opcua/node-opcua #685: Reconnection failure after session lost in LINUX mint

ElintonSilva, 2019-10-14:
**I'm submitting a ...**

 - (x) bug report => 
**Current behavior**
Have diferrents comportaments with the same code in differents Os (windows and linux). After a time disconnected from the server, when trying to reconnect, some client events isn't trigged.

**Expected behavior**
Try to connect with server.

**Minimal reproduction of the problem with instructions**

CODE:

const  { 
  OPCUAClient ,
  resolveNodeId, 
  AttributeIds,
  ClientMonitoredItemGroup, 
  TimestampsToReturn
 } = require("node-opcua-client");

const endpointUri = "opc.tcp://10.10.11.248:4840";

(async () => {

    try {

        const options = {
            connectionStrategy: {
                maxRetry: 2,
                initialDelay: 1000,
                maxDelay: 15000, 
            },
            keepSessionAlive: true,
            endpoint_must_exist: false,
            requestedSessionTimeout: 1000
        };

        const client = OPCUAClient.create(options);

        client.on("backoff", (retry, delay) => {
            console.log("Backoff ", retry, " next attempt in ", delay, "ms");
        });

        client.on("connection_lost", () => {
            console.log("****Connection lost****");
        });

        client.on("connected", () => {
            console.log("****Connected****");
        });

        client.on("connection_failed", () => {
            console.log("****Connection failed****");
        });

        client.on("start_reconnection", () => {
            console.log("****Starting reconnection****");
        });

        client.on("after_reconnection", (err) => {
            console.log("After Reconnection event =>", err); //retorna null
        });

        client.on("reconnection_attempt_has_failed", (err,mess) => {
            console.log("Areconnection_attempt_has_failed =>", err, mess);
        });
        
        /*
        client.on("send_request", (res) => {       
            console.log("Send request =>",res);
        });
        
        client.on("receive_response", (res) => {       
            console.log("Receive response =>", res);
        });
        */

        client.on("abort", () => {
            console.log("****Aborted****");
        });

        console.log("------point 1------");

        await client.connect(endpointUri);

        

        console.log("------point 2------");

        let session = await client.createSession();

        console.log("------point 3------");

        let subscription = await session.createSubscription2({
            maxNotificationsPerPublish: 100,
            publishingEnabled: true,
            requestedLifetimeCount: 100,
            requestedMaxKeepAliveCount: 10,
            requestedPublishingInterval: 100
        });

        console.log("------point 4------");

        subscription.on("internal_error", (n) => {
          console.log(n.toString());
        });

        const itemsToMonitor = [
           {
                attributeId: AttributeIds.Value,
                nodeId: resolveNodeId("ns=4;s=|var|XP315.Application.UserPrg.SPINDLE_1") //variable name
            }
 
        ];

        const optionsGroup = {
            discardOldest: true,
            queueSize: 1,
            samplingInterval: 10
        };
        const monitoredItemGroup = ClientMonitoredItemGroup.create(subscription, itemsToMonitor, optionsGroup, TimestampsToReturn.Both);

        // subscription.on("item_added",function(monitoredItem){
        monitoredItemGroup.on("initialized",  () => {
            console.log(" Initialized !");
        });

         monitoredItemGroup.on("terminated",  () => {
            console.log(" Terminated !");
        });

        monitoredItemGroup.on("changed",  (monitoredItem, dataValue, index) => {
            //console.log("Changed on ", index ,  dataValue.value.toString());
            console.log("Changed on ==>", dataValue.value.toString());
        });

        monitoredItemGroup.on("err",  (message) => {
            console.log("err on ==>", message);
        });

        console.log("------point 5------");

        await new Promise((resolve) => setTimeout(resolve, 100000));

        console.log("------point 6------");

        await monitoredItemGroup.terminate();

        await session.close();
        await client.disconnect();
        console.log("Done !");

    } catch (err) {
        console.log("Error", err);
    }})();
  
Same code runnig in same time, 
<----Linux     |        Windows----->
[image]

  
**What is the motivation / use case for changing the behavior?**
When the server lost connection during 5 min and back, the client can't communicate anymore.

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 

  * (x) I have installed node-opcua as a package ( using npm install )

        
  * Device: PC
  * OS version: 4.15.0-20-generic GNU/Linux
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : Mint
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: PLC
     - Version:XP315
     - Manufacturer/Software vendor:Altus
     - link : https://
     
    
* **node-opcua version:**  :
Version : ^2.1.9

erossignon, 2019-10-17:
Can you describe how you did "When the server lost connection" ....
Can you confirm that reconnection works on Windows but not on LinuxMint ?

ElintonSilva, 2019-10-17:
Hi,

_Can you describe how you did "When the server lost connection"_: 
to reproduce the lost connection, i remove the ethernet cable from the PLC or turn off and then turn on.
_Can you confirm that reconnection works on Windows but not on LinuxMint ?_
Yes, in Windows the reconnection is reestablished, Even after hours trying to connect. But on LinuxMint, after +-10 min,  don´t reconnect anymore.

ElintonSilva, 2019-10-31:
Hi @erossignon ,
with you last update in version 2.2.0 , can't reproduce the issue enymore. 
Thanks for your job.

closed 2019-10-31 by ElintonSilva