node-opcua/node-opcua #691: Warning : your opcua client is sending multiple requests simultaneously to the server

gilesbradshaw, 2019-11-01:
Hi just converting from old version 

Everything is working fine but I am now getting in yellow chalk

```
Warning : your opcua client is sending multiple requests simultaneously to the server CreateSubscriptionRequest
Warning : please fix your application code
```

I wonder if there really is a problem with my application code - it works fine and as far as I can tell your code deals fine with this situation.

Might be good to at least be able to hide this message..

sanlop73, 2019-11-04:
Hi:

I have experienced the same.
I have applied ClientMonitoredItemGroup and the message went away.

const monitoredItemGroup = ClientMonitoredItemGroup.create(subscription, itemsToMonitor, optionsGroup, TimestampsToReturn.Both);

Hope it helps.

erossignon, 2019-11-09:
This warning was added  in 2.2.0 to help  developers that uses node-opcua to detect when multiple concurrents requests are send to a server. Some Server may not accept too many requests at once and could crash or could refuse to serve the request or could cause long delay in processing request if the client  bombards the server with too many request at one.
Please check your code, I can conduct private code review of your code under professional service to help identifying issue and providing alternative ideas for your client.

closed 2020-01-08 by erossignon

pelis, 2020-01-28:
Even though this issue is closed already, I would like to add a few thoughts:
ClientMonitoredItem vs. ClientMonitoredItemGroup

```
ClientMonitoredItem, e.g:
monitoredItem = opcua.ClientMonitoredItem.create(opcSubscription,
                        { nodeId: id, attributeId: AttributeIds.Value}, requestParams, TimestampsToReturn.Both);
```

You may provide requestParams per item, e.g. deadband filter, etc.
Apparently, when creating multiple monitored items in a loop that leades to the warning discussed here. You would probably need need to implement a delayed create loop in that case.
I think it would help if ClientMonitoredItem() could provide a callback, which indicates that you may now create the next monitored item.

```
ClientMonitoredItemGroup, e.g:
ClientMonitoredItemGroup.create(self.opcSubscription, itemsToMonitor, {}, TimestampsToReturn.Both);
```
This makes it easier to create multiple monitored items, however, it seems you may not specify requestParams individually for each item.
It would be nice if, for example, if itemsToMonitor would allow to pass individual requestParams.
e.g.:
```
itemToMonitor: {
     nodeId: NodeId,
     attributeId: attId,
     requestParams: {}
 }
```