node-opcua/node-opcua #693: Node opcua 2.2.0 version:  Cannot find ANONYMOUS user token policy in end point description

samminen, 2019-11-04:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**

(node:3748) UnhandledPromiseRejectionWarning: Error: Cannot find ANONYMOUS user token policy in end point description
    at createAnonymousIdentityToken (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:135:15)
    at OPCUAClientImpl.createUserIdentityToken (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:1253:41)
    at OPCUAClientImpl._activateSession (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:945:14)
    at _createSession (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:429:22)
    at ClientSecureChannelLayer.performMessageTransaction (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:904:13)
    at modified_callback (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1292:31)
    at process_request_callback (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:137:5)
    at ClientSecureChannelLayer._on_message_received (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:745:9)
    at MessageBuilder.ClientSecureChannelLayer.messageBuilder.on (C:\Edge\Solution\edge\rig-connectors\opc-ua-client-service\src\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:387:22)
    at MessageBuilder.emit (events.js:182:13)
warning.js:18
(node:3748) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
warning.js:18
(node:3748) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

**Expected behavior**
When I create subscriptions using Node Opcua 2.2.0 version, I see above "Cannot find ANONYMOUS user token policy in end point description" issue.

**Minimal reproduction of the problem with instructions**

private  userIdentity: NodeOPCUA.UserIdentityInfo /*NodeOPCUA.UserIdentityInfoUserName*/ = {type: NodeOPCUA.UserTokenType.UserName | 
                                                                                                NodeOPCUA.UserTokenType.Anonymous |
                                                                                                NodeOPCUA.UserTokenType.Certificate, 
                                                                                                userName:"", password:""};

this.userIdentity =
                {
                    type: NodeOPCUA.UserTokenType.UserName,
                    userName: this.userName,
                    password: subDecryptedPassrdStr
                };

  1. _I have Node opcus server 2.x version with basic nodes created and updating with random values___
  2. _In OPCUA client (2.2.0) I created monitor item for the node-id and after some time, I see above exception/warning________
  3. ________________
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: ___WIN10__
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =2.2.0

samminen, 2019-11-05:
Is there a sample program where UserIdentity ( AnonymousIdentity | UserIdentityInfoX509 | UserIdentityInfoUserName) options implemented ?

erossignon, 2019-11-09:
@samminen Please provide a complete ts script in a gist that reproduces the issue.

rafaturtle, 2020-02-24:
Hi. I'm having the same issue, using the simple client tutorial againsta a Kepware V6 server on a different server. Firewall is open and I've configured the endpoint to accept security None. It fails when creating the session.

<script src="https://gist.github.com/rafaturtle/68adbd8b5e280d87dcd7045ed07d3cef.js"></script>

youngkzy, 2020-05-20:
Found a similar issue when using "node-red-contrib-opcua" breaks when it was upgraded to use node-opcua 2.4.4.  Use version prior to that, Kepware connects fine using anonymous security None.  Have not tested security levels yet on newer versions.  Usually "No Security" is chosen to first get connection for remote clients.  This busted while also using LOCAL Kepware Server.

erossignon, 2020-05-27:
Can you please run this gist : https://gist.github.com/erossignon/5b2a5ff9edf7abd72d604414def13c2b
for example:
```
ts-node get_endpoints -e opc.tcp//opcuademo.sterfive.com:26543
`````

This will output the exposed endpoints of your server such as 

```
ndpoint                                Application URI                              Product URI       Application Name  securityLevel  Security Mode   securityPolicyUri                                          Type    certificate  discoveryUrls
--------------------------------------  -------------------------------------------  ----------------  ----------------  -------------  --------------  ---------------------------------------------------------  ------  -----------  -------------
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         3              None            http://opcfoundation.org/UA/SecurityPolicy#None            Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         10              Sign            http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15   Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         15              Sign            http://opcfoundation.org/UA/SecurityPolicy#Basic256        Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         20              Sign            http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256  Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         30              SignAndEncrypt  http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15   Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         35              SignAndEncrypt  http://opcfoundation.org/UA/SecurityPolicy#Basic256        Server  ...
opc.tcp://opcuademo.sterfive.com:26543  urn:opcuademo.sterfive.com:NodeOPCUA-Server  NodeOPCUA-Server  NodeOPCUA         50              SignAndEncrypt  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256  Server  ...

Identify Token for : Security Mode= None  Policy= http://opcfoundation.org/UA/SecurityPolicy#None
policyId                    tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
--------------------------  ---------  ---------------  -----------------  ---------------------------------------------------------
username_basic256           1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256
username_basic128           1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
username_basic256Sha256     1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
certificate_basic256Sha256  2          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
certificate_basic256        1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256
certificate_basic128        1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
certificate_basic256Sha256  1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
certificate_basic256Sha256  2          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
anonymous                   0          null             null               null

Identify Token for : Security Mode= Sign  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
policyId          tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
----------------  ---------  ---------------  -----------------  -----------------
usernamePassword  1          null             null               null
certificateX509   2          null             null               null
anonymous         0          null             null               null

Identify Token for : Security Mode= Sign  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256
policyId          tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
----------------  ---------  ---------------  -----------------  -----------------
usernamePassword  1          null             null               null
certificateX509   2          null             null               null
anonymous         0          null             null               null

Identify Token for : Security Mode= Sign  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
policyId          tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
----------------  ---------  ---------------  -----------------  -----------------
usernamePassword  1          null             null               null
certificateX509   2          null             null               null
anonymous         0          null             null               null

Identify Token for : Security Mode= SignAndEncrypt  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
policyId          tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
----------------  ---------  ---------------  -----------------  -----------------
[... cut ...]
Identify Token for : Security Mode= SignAndEncrypt  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
policyId          tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
----------------  ---------  ---------------  -----------------  -----------------
usernamePassword  1          null             null               null
certificateX509   2          null             null               null
anonymous         0          null             null               null

success !!
````

erossignon, 2020-05-30:
@rafaturtle @youngkzy For reference Kepware may prevent unsecure connection and in this situation the client  mus use secure connection and its certificate must be trusted on Kepware side. Mika explained it very well in  https://github.com/mikakaraila/node-red-contrib-opcua/issues/151.

@samminen , please provide additional information to help us understand your specific problem ?

erossignon, 2021-04-24:
@samminen I think that this problem has faded away on your side . Let closed,

closed 2021-04-24 by erossignon

marutimuthu, 2023-10-23:
This error is due to anonymous login being disable by default in Kepserver.

To allow anonymous login: Edit -> Properties -> OPC UA -> Allow Anonymous Login -> Yes

I was facing the same issue and it was resolved after anonymous login was enabled.