node-opcua/node-opcua #697: Node opcua 2.2.0: keepSessionAlive: true is not keeping active connection between client and server

samminen, 2019-11-05:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**
In my opcua client create options, I set "keepSessionAlive: true" and started client. It will connect to opcua server but after opcua server default session timeout (60 sec in my case) opcua client automatically receive connection lost event and it is reconnecting to server immediately. This is observed for every 60 sec. 

**Expected behavior**
As per the documentation, keepSessionAlive: true should keep the active connection between client and server even if there are no active subscriptions from node opcua client.

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: __WIN10___
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =2.2.0

samminen, 2019-11-05:
workaround I tried to avoid client reconnect is below one. getting session timeout from server and creating a temp session and closing it at a certain interval in a loop.

private sessionTimeout: number = 60000;

         this.intervals.push(setInterval( async ()=> {
                try {
                        let tempSession =  await this.client.createSession();
                        this.sessionTimeout = tempSession.timeout;
                        await tempSession.close()
                }catch(err) {

                }
            }, this.sessionTimeout/2));

erossignon, 2019-11-09:
@samminen  please provide a fully operating gist that demonstrate the issue.

samminen, 2019-11-12:
Here is sample opcua client with version 2.2.0 and when i just run this code, it will connect to OPC server and for every 60 sec, I see connection lost and it is triggering below events in opcua client.

If any active subscription exists (if I have active monitoredItems() ) then since session is active, this connection will not break.  below reconnect happens if there is no ACTIVE SESSION.

NOTE: OPCUA server session timeout in this case was 60000ms

opcua server connection lost
start reconnection
reconnect successful !!
opcua server connection lost
start reconnection
reconnect successful !!

import {
    OPCUAClient,
    MessageSecurityMode, SecurityPolicy,
    AttributeIds,
    makeBrowsePath,
    ClientSubscription,
    TimestampsToReturn,
    MonitoringParametersOptions,
    ReadValueIdLike,
    ClientMonitoredItem,
    DataValue,
    ClientSession
    
 } from "node-opcua";
import { Session } from "inspector";

 const options = {
    applicationName: "MyClient",
    keepSessionAlive : true,
    //connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpoint_must_exist: false,
};
const client = OPCUAClient.create(options);
// const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
const endpointUrl = "opc.tcp://NTBK579005.corp.halliburton.com:26543";

client.on('connection_lost', () => {
    console.info("opcua server connection lost");
});

client.on("backoff", async (number, delay) => {
    console.info(`Connection retry attempt #${number} retrying in ${delay/1000.0} seconds`);
});

client.on("start_reconnection", async () => {
    
    console.info("start reconnection");
});
client.on("connection_reestablished", async () => {
    
    console.info("reconnect successful !!");

});
client.on("close", async  () => {
    
    console.info("close and disconnect");

});
client.on("timed_out_request", async (request) => {
    console.info(`Client timeout request Received: ${request.toString()}`);
});
//connect to server
clientConnect();

async function clientConnect(): Promise<void> {
    await client.connect(endpointUrl);
    console.log("connected !");

}

erossignon, 2019-11-12:
* as said in the tin , keepSessionAlive is to keep session alive.
* In your case, there is no session yet to keep alive.

```javascript
async function clientConnect(): Promise<void> {
    await client.connect(endpointUrl);
    console.log("connected !");
    const session = await client.createSession();
    session.on("keepalive", () => {
        console.log("Session is sending a request so that it can stay alive");
    });

}
```

erossignon, 2020-01-09:
@samminen , i guess this helped... Let's close now

closed 2020-01-09 by erossignon