node-opcua/node-opcua #699: npm install node-opcua  certificate error

robinxc, 2019-11-12:
node-opcua-certificate-manager@2.2.0 postinstall /usr/local/windiot/node_modules/node-opcua-certificate-manager
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem

Error: EACCES: permission denied, mkdir '/usr/local/windiot/node_modules/node-opcua-certificate-manager/certificates'
    at Object.mkdirSync (fs.js:823:3)
    at Object.mkdir (/usr/local/windiot/node_modules/node-opcua-pki/lib/pki/toolbox.ts:79:12)
    at /usr/local/windiot/node_modules/node-opcua-pki/lib/crypto_create_CA.ts:211:9
    at cb (util.js:196:39)
    at processTicksAndRejections (internal/process/task_queues.js:80:21)
npm WARN windiot@1.0.0 No description
npm WARN windiot@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-certificate-manager@2.2.0 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-opcua-certificate-manager@2.2.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-12T11_09_39_031Z-debug.log

erossignon, 2019-11-12:
please fill the template

mstfldmr, 2019-11-15:
don't run it as root, or use `npm install node-opcua --unsafe-perm=true`

closed 2019-11-18 by erossignon