node-opcua/node-opcua #700: Can't be installed on Ubuntu 18.04.3

mstfldmr, 2019-11-12:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**
I want to install on a newly upgraded Ubuntu 18.04.3 server with the command:
`npm i node-opcua`

I get error message about node-opcua-certificate-manager:
```

> node-opcua-certificate-manager@2.2.0 postinstall /home/ubuntu/tmp/node_modules/node-opcua-certificate-manager
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem

Error: EACCES: permission denied, mkdir '/home/ubuntu/tmp/node_modules/node-opcua-certificate-manager/certificates'
    at Object.fs.mkdirSync (fs.js:885:18)
    at Object.mkdir (/home/ubuntu/tmp/node_modules/node-opcua-pki/lib/pki/toolbox.ts:79:12)
    at util_1.callbackify (/home/ubuntu/tmp/node_modules/node-opcua-pki/lib/crypto_create_CA.ts:211:9)
    at cb (util.js:1019:39)
    at _combinedTickCallback (internal/process/next_tick.js:139:11)
    at process._tickCallback (internal/process/next_tick.js:181:9)
npm WARN enoent ENOENT: no such file or directory, open '/home/ubuntu/tmp/package.json'
npm WARN tmp No description
npm WARN tmp No repository field.
npm WARN tmp No README data
npm WARN tmp No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-certificate-manager@2.2.0 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-opcua-certificate-manager@2.2.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-12T10_26_59_474Z-debug.log

```

erossignon, 2019-11-13:
what version of nodejs ? which version of npm ? what environment ( raspberry ? docker container ? wsl ? any antivirus software scanning the drive ? ) are they any other processes in action in /home/ubuntu/tmp ?
Can you check permission of /home/ubuntu/tmp/node_modules/node-opcua-certificate-manager/certificates ?
is the drive consistent ? ( run fsck )

Please provide a  more detailed set of step that reproduces the problem and a better description of your environment.

mstfldmr, 2019-11-15:
- I have this problem while running Ubuntu 18.04 on an AWS EC2 instance.
- There is no anti-virus or any other process using the directory.
- Nodejs was installed from the repository as explained at [](https://github.com/nodesource/distributions/blob/master/README.md)
- When I run it as regular user, it installs well.
- I get error only when I run it as **root** without ` --unsafe-perm=true` parameter.

```
$ node --version
v12.13.0

$ npm --version
6.12.0
```

Here is the last lines of the log file:

```
5811 warn tmp5@1.0.0 No description
5812 warn tmp5@1.0.0 No repository field.
5813 verbose stack Error: node-opcua-certificate-manager@2.3.0 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem`
5813 verbose stack Exit status 1
5813 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
5813 verbose stack     at EventEmitter.emit (events.js:210:5)
5813 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
5813 verbose stack     at ChildProcess.emit (events.js:210:5)
5813 verbose stack     at maybeClose (internal/child_process.js:1021:16)
5813 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
5814 verbose pkgid node-opcua-certificate-manager@2.3.0
5815 verbose cwd /home/ubuntu/tmp5
5816 verbose Linux 4.15.0-1051-aws
5817 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "node-opcua"
5818 verbose node v12.13.0
5819 verbose npm  v6.12.0
5820 error code ELIFECYCLE
5821 error errno 1
5822 error node-opcua-certificate-manager@2.3.0 postinstall: `node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem`
5822 error Exit status 1
5823 error Failed at the node-opcua-certificate-manager@2.3.0 postinstall script.
5823 error This is probably not a problem with npm. There is likely additional logging output above.
5824 verbose exit [ 1, true ]
```

erossignon, 2019-11-17:
OK , I have updated the readme file to specify --unsafe-perm=true

closed 2019-11-17 by erossignon