node-opcua/node-opcua #703: Crawler unhandled exception

mstfldmr, 2019-11-15:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**

`const obj = await crawler.read(nodeId);` causes the unhandled 

```
(node:13011) UnhandledPromiseRejectionWarning: Error
    at Object.assert (/home/mustafa/Desktop/Crawler/node_modules/node-opcua-assert/dist/index.js:10:21)
    at _fetch_elements (/home/mustafa/Desktop/Crawler/node_modules/node-opcua-client-crawler/dist/node_crawler.js:42:25)
    at NodeCrawler._resolve_deferred_readNode (/home/mustafa/Desktop/Crawler/node_modules/node-opcua-client-crawler/dist/node_crawler.js:604:42)
    at task4 (/home/mustafa/Desktop/Crawler/node_modules/node-opcua-client-crawler/dist/node_crawler.js:411:22)
    at /home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:2952:28
    at replenish (/home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:444:21)
    at /home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:449:13
    at eachOfLimit$1 (/home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:475:34)
    at awaitable(eachOfLimit$1) (/home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:208:32)
    at eachOfGeneric (/home/mustafa/Desktop/Crawler/node_modules/async/dist/async.js:510:16)
(node:13011) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:13011) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

**Minimal reproduction of the problem with instructions**
```
        const client = await opcua.OPCUAClient.create(
        {
            connectionStrategy: {
                maxRetry: 2,
                initialDelay: 2000,
                maxDelay: 10 * 1000
            },
            endpoint_must_exist: false
        });
        client.on("backoff", () => console.log("retrying connection"));

        await client.connect(endpointUrl);

        const session = await client.createSession();

        const NodeCrawler = opcua.NodeCrawler;
        console.log(NodeCrawler);

        var crawler = await new NodeCrawler(session);

        client.on("send_request", () => {
          console.log('send_request');
        });

        client.on("receive_response", () => {
          console.log('receive_response');
        });

        let t5 = Date.now();

        const obj = await crawler.read(nodeId);
        console.log(" Time        = ", (new Date()).getTime() - t5);
        console.log(" read        = ", crawler.readCounter);
        console.log(" browse      = ", crawler.browseCounter);
        console.log(" browseNext  = ", crawler.browseNextCounter);
        console.log(" transaction = ", crawler.transactionCounter);
        crawler.dispose();
````

**Please tell us about your environment:**
Ubuntu 18.04 gues OS running in Virtualbox (6.0) on Mac (10.14.6) host.

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : Ubuntu 18.04
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: opcua-asyncio
     - Version: 0.6.1
     - Manufacturer/Software vendor:_________
     - link : https://github.com/FreeOpcUa/opcua-asyncio.git
     
    
* **node-opcua version:**  :
      "version": "2.3.0",
      "resolved": "https://registry.npmjs.org/node-opcua/-/node-opcua-2.3.0.tgz",
      "integrity": "sha512-u5h1mk/NlokyOP1zEJfXiSl5qc6N633TIi/nXNVHDcJl5gosMLsCjrljl2s6RmGjWDiZXEGBB8H6U0x+CdUxog==",

* **Node:** 
   `node --version` = v12.13.0

erossignon, 2019-11-16:
Can you specify what is nodeId ?

Wensh, 2020-07-24:
@erossignon I have the same issue and my nodeID is "ns=3;s=NotesCollection"

erossignon, 2020-10-02:
OK , I cannot reproduce unless you provide access specify code for the server;

I ran https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-minimal.py
and this code : 
<description>
<summary> script </summary>
<details>

```typescript
const opcua = require("node-opcua");

const nodeId = "ns=0;i=88";
(async () => {

    const endpointUrl = "opc.tcp://localhost:4840";

    try {

        const client = await opcua.OPCUAClient.create(
            {
                connectionStrategy: {
                    maxRetry: 2,
                    initialDelay: 2000,
                    maxDelay: 10 * 1000
                },
                endpoint_must_exist: false
            });
        client.on("backoff", () => console.log("retrying connection"));

        await client.connect(endpointUrl);

        const session = await client.createSession();

        const NodeCrawler = opcua.NodeCrawler;
        console.log(NodeCrawler);

        var crawler = await new NodeCrawler(session);

        client.on("send_request", () => {
            console.log('send_request');
        });

        client.on("receive_response", () => {
            console.log('receive_response');
        });

        let t5 = Date.now();

        const obj = await crawler.read(nodeId);
        console.log(" Time        = ", (new Date()).getTime() - t5);
        console.log(" read        = ", crawler.readCounter);
        console.log(" browse      = ", crawler.browseCounter);
        console.log(" browseNext  = ", crawler.browseNextCounter);
        console.log(" transaction = ", crawler.transactionCounter);
        crawler.dispose();
    }
    catch (err) {
        console.log("err=", err);
    }
})();
```
</details>
</description>

and it produces 

<description>
<summary>output</summary>
<details>
[Function: NodeCrawler]
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
send_request
receive_response
 Time        =  1435
 read        =  8822
 browse      =  1397
 browseNext  =  0
 transaction =  22
send_request
send_request
receive_response
send_request
receive_response
send_request
send_request
send_request
send_request
send_request
send_request
send_request
receive_response
send_request
receive_response
send_request
send_request
send_request
send_request
send_request
</details>
</description>
Can you point me out to a sample python script that exhibits the issue ?

erossignon, 2021-02-07:
Closing due to lack of activities in this thread.

closed 2021-02-07 by erossignon