node-opcua/node-opcua #704: Server will not start

blackknight36, 2019-11-19:
I'm attempting to run this module on CentOS 7 however the server fails to start after installing from source as documented in the README file.

The `node packages/node-opcua-samples/bin/simple_server` command fails with the following message.

```
internal/modules/cjs/loader.js:895
    throw err;
    ^

Error: Cannot find module 'node-opcua'
Require stack:
- /usr/src/node-opcua/packages/node-opcua-samples/bin/simple_server.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:892:15)
    at Function.Module._load (internal/modules/cjs/loader.js:785:27)
    at Module.require (internal/modules/cjs/loader.js:956:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/usr/src/node-opcua/packages/node-opcua-samples/bin/simple_server.js:7:15)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1103:10)
    at Module.load (internal/modules/cjs/loader.js:914:32)
    at Function.Module._load (internal/modules/cjs/loader.js:822:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1143:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/usr/src/node-opcua/packages/node-opcua-samples/bin/simple_server.js'
  ]
}
```
I am not certain how to fix this, the module definitely does exist.

erossignon, 2019-11-23:
Can you provide the output of the `npm install` step ?

blackknight36, 2019-11-25:
@erossignon Sure thing.  The output is quite long, I've created a gist for it at https://gist.github.com/blackknight36/15941a652bd316b23b59b8f328140e84

erossignon, 2019-11-26:
The probleme seems to be that the postinstall step has failed with this error for some reason
```` 
npm WARN lifecycle node-opcua@2.0.0~postinstall: cannot run in wd %s %s (wd=%s) node-opcua@2.0.0 npm run bootstrap && npm run build && npm run generate /usr/src/node-opcua
```` 

can you try to add the `--unsafe-perm` parameter during install ?

```` 
npm install --unsafe-perm
```` 

see https://stackoverflow.com/questions/18136746/npm-install-failed-with-cannot-run-in-wd

blackknight36, 2019-12-05:
@erossignon Sorry for the delay.  Using the unsafe-perm option doesn't help, I'm still getting a syntax error as shown below.

```
[root@hostname node-opcua]# npm install --unsafe-perm

> node-opcua@2.0.0 postinstall /usr/src/node-opcua
> npm run bootstrap && npm run build && npm run generate

> node-opcua@2.0.0 bootstrap /usr/src/node-opcua
> node  ./node_modules/lerna/cli.js bootstrap --hoist --reject-cycles --sort --concurrency=1

lerna notice cli v3.19.0
lerna info Bootstrapping 73 packages
lerna info Installing external dependencies
lerna info hoist Finished bootstrapping root
lerna info hoist Pruning hoisted dependencies
lerna info hoist Finished pruning hoisted dependencies
lerna info Symlinking packages and binaries
lerna info lifecycle node-opcua-certificate-manager@2.4.0~postinstall: node-opcua-certificate-manager@2.4.0

> node-opcua-certificate-manager@2.4.0 postinstall /usr/src/node-opcua/packages/node-opcua-certificate-manager
> node test_helpers/create_certificates.js certificate -s -o certificates/client_selfsigned_cert_2048.pem

/usr/src/node-opcua/node_modules/chalk/source/index.js:103
        ...styles,
        ^^^

SyntaxError: Unexpected token ...
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:549:28)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)                                                                                                                             
    at Object.<anonymous> (/usr/src/node-opcua/node_modules/node-opcua-pki/lib/crypto_create_CA.ts:30:1)                                                              
lerna info lifecycle node-opcua-certificate-manager@2.4.0~postinstall: Failed to exec postinstall script                                                              
lerna ERR! lifecycle "postinstall" errored in "node-opcua-certificate-manager", exiting 1                                                                             
                                                                                                                                                                      
npm ERR! Linux 3.10.0-1062.4.1.el7.x86_64
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "bootstrap"
npm ERR! node v6.17.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! node-opcua@2.0.0 bootstrap: `node  ./node_modules/lerna/cli.js bootstrap --hoist --reject-cycles --sort --concurrency=1`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-opcua@2.0.0 bootstrap script 'node  ./node_modules/lerna/cli.js bootstrap --hoist --reject-cycles --sort --concurrency=1'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-opcua package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node  ./node_modules/lerna/cli.js bootstrap --hoist --reject-cycles --sort --concurrency=1
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-opcua
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-opcua
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/node-opcua/npm-debug.log

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.2.7 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN acorn-jsx@5.1.0 requires a peer of acorn@^6.0.0 || ^7.0.0 but none was installed.
npm ERR! Linux 3.10.0-1062.4.1.el7.x86_64
npm ERR! argv "/usr/bin/node" "/bin/npm" "install" "--unsafe-perm"
npm ERR! node v6.17.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! node-opcua@2.0.0 postinstall: `npm run bootstrap && npm run build && npm run generate`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-opcua@2.0.0 postinstall script 'npm run bootstrap && npm run build && npm run generate'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-opcua package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm run bootstrap && npm run build && npm run generate
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-opcua
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-opcua
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/node-opcua/npm-debug.log
```

erossignon, 2019-12-05:
OK; , we are getting there !
see in the log :
`npm ERR! node v6.17.1`

you are using a very old version of nodejs that some package we are relying on are not supporting anymore .

Please make sure to use node v8.1.0 minium , v12 prefered

mtrabelsi, 2020-02-17:
i had the same issue, after fixing that issue, i had new issue ( tested on Mac os and Ubuntu 18 ) it is complaining about the certificate. Are you sure the documentation is correct? ( it doesn't as it seem to claim) !

blackknight36, 2020-02-17:
> you are using a very old version of nodejs that some package we are relying on are not supporting anymore .

Unfortunately that is a fact of life on CentOS.  The node binary is provided by the nodejs package which is version 6.17.1.

```
user@dev99:~$ rpm -qi nodejs
Name        : nodejs
Epoch       : 1
Version     : 6.17.1
Release     : 1.el7
Architecture: x86_64
Install Date: Thu 24 Oct 2019 02:09:37 PM EDT
Group       : Development/Languages
Size        : 17163981
License     : MIT and ASL 2.0 and ISC and BSD
Signature   : RSA/SHA256, Tue 18 Jun 2019 01:14:09 PM EDT, Key ID 6a2faea2352c64e5
Source RPM  : nodejs-6.17.1-1.el7.src.rpm
Build Date  : Tue 18 Jun 2019 08:50:52 AM EDT
Build Host  : buildvm-18.phx2.fedoraproject.org
Relocations : (not relocatable)
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : http://nodejs.org/
Bug URL     : https://bugz.fedoraproject.org/nodejs
Summary     : JavaScript runtime
Description :
Node.js is a platform built on Chrome's JavaScript runtime
for easily building fast, scalable network applications.
Node.js uses an event-driven, non-blocking I/O model that
makes it lightweight and efficient, perfect for data-intensive
real-time applications that run across distributed devices.
```

If this package requires a newer version of node that is fine however many CentOS users will run into the same problem.

erossignon, 2020-05-30:
@blackknight36  I am sorry, node@6 is no longer supported and we wont fix this. please use alternative ways to install a more recent version of nodejs.

closed 2020-05-30 by erossignon