node-opcua/node-opcua #705: Cannot Recreate Session

seang96, 2019-11-21:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**
I suspect the issue occurs when the S7-1500 device is downloaded to and initializes again. I get a message "recreate_session_and_reperform_transaction => Already in Progress" and my requests all error with "cannot recreate session". It has been in this state for over half of a day. This seems to occur after the error "publushEngine: invalid state" (I assume there is a typo and this should be publishEngine too). This has occurred on both my development machine and on a production server.

**Expected behavior**
The session is recreated.

**Minimal reproduction of the problem with instructions**

  1. Download to S7-1500 PLC
  2. An error "publushEngine: invalid state" occurs
  3. Attempts to recreate session and it never succeeds

gist: Any client connected to the PLC should do

  
**What is the motivation / use case for changing the behavior?**
I need the session to recover to keep my application alive.

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: IPC547G
  * OS version:  Windows Server 2016 Standard with Hyper-V
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : NixOS 19.09
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:S7-1500
     - Version:V2.5.2
     - Manufacturer/Software vendor:Siemens
     - link : https://new.siemens.com/global/en/products/automation/systems/industrial/plc/simatic-s7-1500.html
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =2.2.0

xmariopereira, 2020-02-20:
Hi @seang96 i had this problem because OPC UA credentials in OPC UA Server were incorrect and Allow Anonymus was off too.

Check credentials!

seang96, 2020-02-20:
@xmariopereira, I do not believe this is my issue on this case since no security and enable guest authentication is checked. Also, read/write/monitor operations all work as well I imagine this wouldn't be possible if credentials were not configured correctly all the time.

xmariopereira, 2020-02-26:
Ok, In my case, in order to reproduce this error I did the following:

1 - OPC UA Client was passing credentials but empty.
2 - No Security and No Authentication were enabled in OPC UA Server.

erossignon, 2020-03-22:
Should be fixed with node-opcua@2.5.1 

@seang96  @xmariopereira  can you please confirm ?

xmariopereira, 2020-03-30:
Hi @erossignon i will test, i will let you know! Thanks!

seang96, 2020-03-30:
> 
> 
> Should be fixed with node-opcua@2.5.1
> 
> @seang96 @xmariopereira can you please confirm ?

The issue seems to be resolved on my end. Thank you!

erossignon, 2020-05-30:
@seang96  thank you,

closed 2020-05-30 by erossignon