node-opcua/node-opcua #710: Is there a sample node opcua client with X.509 certificates ?

samminen, 2019-12-17:
I am looking for a sample node opcua client implementation which uses x.509 certs to connect to OPCUA server. Can anyone help me with sample code ? what are the important properties need to fill for x.509 in client options and create session case ? 

Appreciate all your help.

erossignon, 2020-01-02:
Hi  @samminen Samminen.
Chapter 8 of my book  provides a step by step example of how to do this

8.4 Create a client session with a X509 user certificate

see https://leanpub.com/node-opcuabyexample

Wensh, 2020-01-13:
Hi @erossignon,

Unfortunately the book only describes creating a default certificate with some test values (nodeopcua-test etc.) and it's using command lines to create the certificates.

Is there a way to create certificates dynamically and programmatically, based on user input? Similar to how UaExpert does it?

Appreciate your help

closed 2020-06-12 by erossignon