node-opcua/node-opcua #711: With 100 item nodes OPCUA client is not working and nodered is getting crashed

akvish, 2019-12-23:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**

I was trying to connect 100 item nodes with a client node. It was throwing an error message : "Error: Transaction has timed out ( timeout = 60000 ms)". After that the complete nodered is crashing.

**Expected behavior**

Connect to the node prorperly and print/write the values

**Minimal reproduction of the problem with instructions**

  1. Create a flow with 100 item nodes connected to one client node. Read the data in an interval of 1s

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

Intel pentium with 128 gb ssd and 4gb ram running in debian 9
  * (X ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( X) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - (X) node-red
       - ( ) other : please specify 
        
  * Device: IPC 227e (Siemens)
  * OS version: Debian 9
    
    - ( ) Windows : version : _________
    - (x ) Linux   : version : Debian 9
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:s7 plc
     - Version:1500 plc
     - Manufacturer/Software vendor:Siemens
     - link : https://
     
    
* **node-opcua version:**  :
 OPC-UA node 0.2.49

* **Node:** 
   `node --version` = 0.20.3

mikakaraila, 2019-12-23:
Have you tested longer interval like 10s? Single read can take longer time so it be overflow with timeout. I will implement after Christmas another way to read multiple items at one request. Another way is to use subscription.

mikakaraila, 2019-12-29:
New feature is now available:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/127

erossignon, 2020-01-02:
closing in favour of mikakaraila/node-red-contrib-opcua#127

closed 2020-01-02 by erossignon