node-opcua/node-opcua #712: Error using node-opcua

ilya211993, 2019-12-30:
Hello,

I have following issue: 

I want to implement my a Server Client to my vue.js-application using node-opcua. For this purpose, I could succesfully install node-opcua (Version: 2.4.1) to my project. However, as soon as I try to import or use the OPCUAClient in order to connect to a server, I get following errors: 

ERROR in ./node_modules/node-opcua-pki/dist/lib/misc/install_prerequisite.js
Module not found: Error: Can't resolve 'child_process' in 'C:\Users\Ilya\Desktop\ATP Aufgabe\node_modules\node-opcua-pki\dist\lib\misc'
 @ ./node_modules/node-opcua-pki/dist/lib/misc/install_prerequisite.js 29:22-46
 @ ./node_modules/node-opcua-pki/dist/lib/pki/toolbox.js
 @ ./node_modules/node-opcua-pki/dist/lib/index.js
 @ ./node_modules/node-opcua-certificate-manager/dist/certificate_manager.js
 @ ./node_modules/node-opcua-certificate-manager/dist/index.js
 @ ./node_modules/node-opcua/dist/server-stuff.js
 @ ./node_modules/node-opcua/dist/index.js
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib??vue-loader-options!./src/views/Home.vue?vue&type=script&lang=js&
 @ ./src/views/Home.vue?vue&type=script&lang=js&
 @ ./src/views/Home.vue
 @ ./src/router/index.js
 @ ./src/main.js

Does anyone know, how I can solve it?

erossignon, 2020-01-02:
node-opcua can only be used on the server side  and will not operate inside the browser.

erossignon, 2020-01-02:
duplicate #194

closed 2020-01-02 by erossignon

zemirco, 2020-03-09:
Hey,

I made our OPC UA client for the browser open source https://github.com/HBM/opcua. It is work in progress but activating a session already works. Maybe it's an option for you.

Best regards,
Mirco