node-opcua/node-opcua #717: Crawler: Cannot get nodeClass of every node

aljoshakoecher, 2020-01-10:
**I'm submitting a ...**

 - (x) bug report
 - ( ) feature request
 - (x) support request

**Current behavior**
Using the node-crawler, I cannot get the nodeClass of every node. I only get the nodeClass of the root node I'm crawling.

**Expected behavior**
I'd like to get every node's nodeClass. I'm not sure if this is a bug or if I'm just missing to pass any options to the crawler. What I'm doing is basically just:

```
const crawler = new NodeCrawler(mySession);
const nodeId = "ObjectsFolder";
crawler.read(nodeId, function (err, obj) {
  console.log(obj);
}
```
And it works, I'm able to crawl different servers (on a PLC, the public Milo example server and another one I wrote with Milo). But I don't get the nodeClasses...
Is there anything I'm missing or is this a bug?

My result currently looks like this:
```
{
    "browseName": "Objects",
    "nodeId": "ns=0;i=85",
    "nodeClass": "1",
    "organizes": [{
        "browseName": "DeviceSet",
        "nodeId": "ns=2;i=5001",
        "typeDefinition": "BaseObjectType",
	"organizes": [{
	    "browseName": "CODESYS Control Win V3 x64",
	    "nodeId": "ns=4;s=|plc|CODESYS Control Win V3 x64",
	    "organizes": [{
                ...
```

Thanks for any help in advance!

aljoshakoecher, 2020-01-16:
Is there anyone who has experienced this too? Or has anyone managed to get every node's nodeClass with the crawler?

erikholleboom, 2020-02-01:
I see similar issue with the value of a variable node. In random they appear or not. 
Running both server and client with node-opcua

closed 2020-02-04 by erossignon

aljoshakoecher, 2020-02-10:
Thanks for trying to fix this, @erossignon! 
I get the value of every node now, so it seems like @erikholleboom's problem was fixed. Unfortunately, I'm still having trouble with the nodeClass.

Tested with v.2.5.0-alpha.6 in combination with a soft-plc on localhost and the public milo example server. Please let me know if I can provide any additional info.

commit c4a516bb (2020-02-04): fix #717 add missing nodeClass property in crawler