node-opcua/node-opcua #720: BrowseFilter isn't working Correct

andi00, 2020-01-15:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**
In the browseFilterFunction the SessionContext is always undefined.

**Expected behavior**
The SessionContext is set.

**Minimal reproduction of the problem with instructions**

Start Server
Connect with Client to Server.
Browse to Folder Test

gist:
import * as opcua from "node-opcua";

const server = new opcua.OPCUAServer({
  port: 4841
});

interface MyAddOptions extends opcua.AddBaseNodeOptions {
  browseFilter?: (session: opcua.SessionContext) => boolean;
}

function browseFilter(context: opcua.SessionContext): boolean {
  if (context == null) {
    return false;
  }
  return true;
}

server
  .start()
  .then(() => {
    const addrSpace = server.engine.addressSpace;
    if (addrSpace === null) return;
    const namespace = addrSpace.getOwnNamespace();
    const options: MyAddOptions = {
      browseName: "Test",
      browseFilter: browseFilter
    };
    namespace.addFolder(addrSpace.rootFolder.objects, options);
  })
  .catch(reason => {
    console.error(reason);
  });
  
**What is the motivation / use case for changing the behavior?**
The motiviation is to Hide some Nodes for some User

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( x ) Windows : version : WIN 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
     
    
* **node-opcua version:**  :  2.4.1

* **Node:** 
   `node --version` = 10.15.3

erossignon, 2020-02-02:
filtering browse based on context is currently not support. Sponsor welcome.

tmx976, 2020-03-19:
For my needs, it has been enough to modify 
**node_modules\node-opcua-server\dist\opcua_server.js:1570**

from
```js
results = server.engine.browse(request.nodesToBrowse);
```
to
```js
results = server.engine.browse(request.nodesToBrowse, session);
```

closed 2020-10-11 by erossignon