node-opcua/node-opcua #724: OPCUA Client failing

bmshukla, 2020-01-29:
**I'm submitting a ...**  (check one with "x")

 - (X) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**

 admin0@mcfdev:~/OPCUA_Client0$ ts-node OPCUA-Client.ts

/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:421
    return new TSError(diagnosticText, diagnosticCodes)
           ^
TSError: ⨯ Unable to compile TypeScript:
OPCUA-Client.ts:106:20 - error TS1252: Function declarations are not allowed inside blocks in strict mode when targeting 'ES3' or 'ES5'. Modules are automatically in strict mode.

106     async function timeout(ms: number) {
                       ~~~~~~~
OPCUA-Client.ts:12:8 - error TS2307: Cannot find module 'node-opcua'.

12 } from "node-opcua";
          ~~~~~~~~~~~~

    at createTSError (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:421:12)
    at reportTSError (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:425:19)
    at getOutput (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:530:36)
    at Object.compile (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:735:32)
    at Module.m._compile (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:814:43)
    at Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Object.require.extensions.<computed> [as .ts] (/usr/local/node-v12.14.0-linux-x64/lib/node_modules/ts-node/src/index.ts:817:12)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
(base) admin0@mcfdev:~/OPCUA_Client0$
**Expected behavior**

Client should run normally.
**Minimal reproduction of the problem with instructions**

  1. _It seems to be a bug_______________
  2. ________________
  3. ________________
  ...

gist: It is showing that 
"OPCUA-Client.ts:12:8 - error TS2307: Cannot find module 'node-opcua'."

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( X) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: Linux Ubuntu 18.04.03_____
    
    - ( ) Windows : version : _________
    - (X ) Linux   : version : _Ubuntu 18.04.03________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  : 2.4.4

* **Node:** 
   `node --version` = v12.14.0

erossignon, 2020-01-29:
#### error 1:
````
OPCUA-Client.ts:106:20 - error TS1252: Function declarations are not allowed inside blocks in strict mode when targeting 'ES3' or 'ES5'. Modules are automatically in strict mode.
106     async function timeout(ms: number) {
````
move timeout function to outer scope

##### error 2:
OPCUA-Client.ts:12:8 - error TS2307: Cannot find module 'node-opcua'.

Looks like node-opcua is not installed 
It should be present in the node_modules folders 

12 } from "node-opcua";

bmshukla, 2020-01-29:
After executing "$ npm install node-opcua-client --save" command  , following packages have been installed in node_modules directory. In this list node-opcua is missing. Now what to do?
===
aggregate-error                 node-opcua-chunkmanager
ansi-regex                      node-opcua-client
ansi-styles                     node-opcua-client-dynamic-extension-object
any-promise                     node-opcua-common
array-flatten                   node-opcua-constants
array-union                     node-opcua-crypto
asn1                            node-opcua-data-model
async                           node-opcua-data-value
backoff                         node-opcua-date-time
balanced-match                  node-opcua-debug
better-assert                   node-opcua-enum
bomstrip                        node-opcua-extension-object
bonjour                         node-opcua-factory
brace-expansion                 node-opcua-generator
braces                          node-opcua-guid
buffer-crc32                    node-opcua-hostname
buffer-from                     node-opcua-nodeid
buffer-indexof                  node-opcua-numeric-range
byline                          node-opcua-object-registry
callbackify                     node-opcua-packet-analyzer
callsite                        node-opcua-packet-assembler
camelcase                       node-opcua-pki
chalk                           node-opcua-pseudo-session
clean-stack                     node-opcua-schemas
cli-spinner                     node-opcua-secure-channel
cli-table                       node-opcua-service-browse
cliui                           node-opcua-service-call
color-convert                   node-opcua-service-discovery
color-name                      node-opcua-service-endpoints
colors                          node-opcua-service-filter
concat-map                      node-opcua-service-history
ctype                           node-opcua-service-query
decamelize                      node-opcua-service-read
deep-equal                      node-opcua-service-register-node
define-properties               node-opcua-service-secure-channel
del                             node-opcua-service-session
delayed                         node-opcua-service-subscription
dir-glob                        node-opcua-service-translate-browse-path
dns-equal                       node-opcua-service-write
dns-packet                      node-opcua-status-code
dns-txt                         node-opcua-transport
emoji-regex                     node-opcua-types
env-paths                       node-opcua-utils
es-abstract                     node-opcua-variant
es-to-primitive                 node-opcua-xml2json
fast-glob                       object.assign
fastq                           object-inspect
fd-slicer                       object-is
fill-range                      object-keys
find-up                         once
foreachasync                    path-exists
fs.realpath                     path-is-absolute
function-bind                   path-type
get-caller-file                 pend
glob                            picomatch
globby                          p-limit
glob-parent                     p-locate
graceful-fs                     p-map
has                             precond
has-flag                        prettier
has-symbols                     progress
hexy                            p-try
ignore                          regexp.prototype.flags
indent-string                   require-directory
inflight                        require-main-filename
inherits                        reusify
ip                              rimraf
is-arguments                    run-parallel
is-callable                     safe-buffer
is-date-object                  set-blocking
is-extglob                      slash
is-fullwidth-code-point         source-map
is-glob                         source-map-support
is-number                       ssh-key-to-pem
is-path-cwd                     string.prototype.trimleft
is-path-inside                  string.prototype.trimright
is-regex                        string-width
is-symbol                       strip-ansi
jsrsasign                       supports-color
locate-path                     thenify
long                            thunky
ltx                             to-regex-range
merge2                          tunnel
micromatch                      @types
minimatch                       typescript
minimist                        underscore
mkdirp                          walk
multicast-dns                   wget-improved
multicast-dns-service-types     which-module
@nodelib                        wrap-ansi
node-opcua-assert               wrappy
node-opcua-basic-types          y18n
node-opcua-binary-stream        yargs
node-opcua-buffer-utils         yargs-parser
node-opcua-certificate-manager  yauzl
=========

bmshukla, 2020-01-29:
it seems that while executing "$ npm install node-opcua-client --save" command , it does not install node-opcua package. So please let me know the command to do so.

erossignon, 2020-01-30:
Yes!  use 
$ npm install node-opcua --save

closed 2020-01-30 by bmshukla

bmshukla, 2020-02-04:
I have executed 

I got the following error.
===
(base) admin0@mcfdev:~/OPCUA_Client0$ ts-node  OPCUA-Client.ts
An error has occured :  Error: The connection cannot be established with server opc.tcp://localhost:26543 .
Please check that the server is up and running or your network configuration.
Err = (connect ECONNREFUSED 127.0.0.1:26543)
    at /home/admin0/OPCUA_Client0/node_modules/node-opcua-client/source/private/client_base_impl.ts:648:27
    at /home/admin0/OPCUA_Client0/node_modules/node-opcua-client/source/private/client_base_impl.ts:583:17
    at /home/admin0/OPCUA_Client0/node_modules/async/dist/async.js:2960:19
    at wrapper (/home/admin0/OPCUA_Client0/node_modules/async/dist/async.js:273:20)
    at iterateeCallback (/home/admin0/OPCUA_Client0/node_modules/async/dist/async.js:418:21)
    at /home/admin0/OPCUA_Client0/node_modules/async/dist/async.js:326:20
    at /home/admin0/OPCUA_Client0/node_modules/async/dist/async.js:2958:17
    at /home/admin0/OPCUA_Client0/node_modules/node-opcua-client/source/private/client_base_impl.ts:550:21
    at /home/admin0/OPCUA_Client0/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:536:28
    at ClientSecureChannelLayer._backoff_completion (/home/admin0/OPCUA_Client0/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1063:17)
==

My OPCUA server is running on port no 4334, but OPCUA client program is  trying to contact on port no.  26543. I am unable to understand this.