node-opcua/node-opcua #728: Cannot read property 'buildInfo'

migu3l705, 2020-02-11:
**Current behavior**

I'm running the git clone example https://github.com/node-opcua/node-opcua-sampleserver.git

When executing: npm start, an error occurs

**Describe the bug**
I get the following error:

C: \ Users \ migu3 \ node-opcua-sampleserver \ node_modules \ node-opcua-server \ dist \ opcua_server.js: 543
        return this.engine.buildInfo;
                           ^

TypeError: Cannot read property 'buildInfo' of undefined
    at OPCUAServer.get buildInfo [as buildInfo] (C: \ Users \ migu3 \ node-opcua-sampleserver \ node_modules \ node-opcua-server \ dist \ opcua_server.js: 543: 28)
    at Object. <anonymous> (C: \ Users \ migu3 \ node-opcua-sampleserver \ server.js: 16: 8)
    at Module._compile (internal / modules / cjs / loader.js: 1157: 30)
    at Object.Module._extensions..js (internal / modules / cjs / loader.js: 1177: 10)
    at Module.load (internal / modules / cjs / loader.js: 1001: 32)
    at Function.Module._load (internal / modules / cjs / loader.js: 900: 14)
    at Function.executeUserEntryPoint [as runMain] (internal / modules / run_main.js: 74: 12)
    at internal / main / run_main_module.js: 18: 47
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-sampleserver@0.0.56 start: `node server.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-opcua-sampleserver@0.0.56 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! C: \ Users \ migu3 \ AppData \ Roaming \ npm-cache \ _logs \ 2020-02-11T20_18_35_164Z-debug.log

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/node-opcua/node-opcua-sampleserver.git
2. cd node-opcua-sampleserver
3. npm install
4. npm start

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: Windows
    
    - ( ) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :

erossignon, 2020-02-12:
The [node-opcua-sampleserver](https://github.com/node-opcua/node-opcua-sampleserver.git) is quite old-date and need to be updated. 

In the meantime, if you want up to date samples for using node-opcua , you can refer to my [book node-opcua by example](https://leanpub.com/node-opcuabyexample/).

It comes with a comprehensive set of step by step examples to create client or server for OPCUA and is regularly updated.

erossignon, 2020-03-22:
I have  updated the node-opcua-sampleserver

closed 2020-03-22 by erossignon