node-opcua/node-opcua #729: Replace native dependency in 0.7.4 (opcua-hostname)

lal12, 2020-02-12:
This is related to 0.7.4. I know it is not the current version, so I am not sure if it is still supported, but node-red-contrib-opcua depends on it, which I guess could be called an "important" project. There is already an issue about using node-opcua@2.x.x (https://github.com/mikakaraila/node-red-contrib-opcua/issues/114).

I would like to use opcua on a system without a build environment. (Actually node-red-opcua). However there is only one depedency which is needed to be build, which is deasync. It is required by node-fqdn which is required by node-opcua-hostname. 

**Describe the solution you'd like**
This is not really an important module, it could easily be replaced by another one without a native dependency.

**Describe alternatives you've considered**
The new code of opcua-hostname could be backported or node-fqdn could be replaced by get-fqdn.

erossignon, 2020-02-18:
using node-opcua@2 in node-red-contrib-opcua is on the way and should be the preferred route.
Because deasync is  Sync and other similar modules requires async behavior, backporting the removal of deasync  is 0.7.4 will be complex task that could be made under professional service by sterfive 
Anyway sponsors are also welcome to help us staying motivated ! https://github.com/sponsors/node-opcua

lal12, 2020-02-18:
>  Because deasync is Sync and other similar modules requires async behavior, backporting the removal of deasync is 0.7.4 will be complex task that could be made under professional service by sterfive

As far as I seen it, removing deasync for this specific module would not change anything, but I am working on porting node-red-contrib-opcua to v2. If this does not work as expected I will do some further investigations into the deasync removal.

erossignon, 2020-06-13:
Dependency to desasync has been removed a long time ago. We can close this.

closed 2020-06-13 by erossignon