node-opcua/node-opcua #734: Missing 'xml-writer' module in node-opcua-xml2json package

mibux, 2020-02-18:
**Current behavior**
After creating a fresh project, installing node-opcua-client package and creating a opcua client, I get a runtime error:

internal/modules/cjs/loader.js:984
  throw err;
  ^

Error: Cannot find module 'xml-writer'

This is valid for version 2.4.4 as well as for 2.5.0-alpha.10

**Describe the bug**
It seems the xml-writer package is missing in the package.json of node-opcua-xml2json package

**To Reproduce**
Steps to reproduce the behavior:
1. create new folder and cd into it
2. => npm init
3. => npm install node-opcua-client
4. create new file (in my case main.js) with following content:

const node_opcua_client_1 = require("node-opcua-client");
const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: Infinity,
};
const opcClientOptions = {
    applicationName: 'myClient',
    connectionStrategy: connectionStrategy,
    securityMode: node_opcua_client_1.MessageSecurityMode.None,
    securityPolicy: node_opcua_client_1.SecurityPolicy.None,
    endpoint_must_exist: false,
};
const client = node_opcua_client_1.OPCUAClient.create(opcClientOptions);

5. => node ./main.js

**Expected behavior**
There should be no error regarding xml-writer module.

 * (X) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC x86
  * OS version: Windows
    
    - (X) Windows : version : Windows 7 Professional
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: S7-1500
     - Version:_________
     - Manufacturer/Software vendor: Siemens
     - link : https://mall.industry.siemens.com/mall/en/de/Catalog/Product/6ES7511-1AK02-0AB0
     

* **node-opcua version:**  :
- Version 2.4.4
- Version 2.5.0-alpha.10

* **Node:** 
   `node --version` = v12.16.0

Adding xml-writer module through "npm install xml-writer" solves this problem.

erossignon, 2020-02-18:
workaround whilst waiting for next release => install xml-writer in your project

erossignon, 2020-03-20:
This should be fix in version 2.5.1. @mibux  can you confirm ?

mibux, 2020-03-20:
Test setup works like a charm. I will integrate it in my bigger project next week and let you know if there are still some issues. Thanks for your work @erossignon

closed 2020-03-20 by erossignon