node-opcua/node-opcua #737: Error when reading asynchronous Variable of a device

mtrabelsi, 2020-02-27:
**Current behavior**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. create a simple server that contains 1 device and 1 variable and return simple string synchronously 
2. change the same server with returning the same variable in an asynchronous way
3. you will get an error

**Expected behavior**
Should support Asychnrounous fetching

**Gist**
```

/*global require,setInterval,console */
const opcua = require("node-opcua");
const mamHelper =  require("./mam")
// Let's create an instance of OPCUAServer
const server = new opcua.OPCUAServer({
    port: 4334, // the port of the listening socket of the server
    resourcePath: "/test", // this path will be added to the endpoint resource name
     buildInfo : {
        productName: "MySampleServer1",
        buildNumber: "7658",
        buildDate: new Date(2014,5,2)
    }
});

function post_initialize() {
    console.log("initialized");
    function construct_my_address_space(server) {

        const addressSpace = server.engine.addressSpace;
        const namespace = addressSpace.getOwnNamespace();

        // declare a new object
        const device = namespace.addObject({
            organizedBy: addressSpace.rootFolder.objects,
            browseName: "IOTA_GATEWAY_DEVICE"
        });

        namespace.addVariable({
            componentOf: device,
            nodeId: "ns=1;s=iota-mam-writer-reader",
            browseName: "mam reader",
            dataType: "String",
            value: {
                refreshFunc: async function (callback) {
                    setTimeout(()=>{
                        callback(
                            null, 
                            new opcua.Variant({dataType: opcua.DataType.String, value: "reply after 1s" })
                        )
                    }, 1000)
                },
                set: function (variant) {
                    const variable2 = parseFloat(variant.value);
                    return opcua.StatusCodes.Good;
                }
            }
        });
    }
    construct_my_address_space(server);
    server.start(function() {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = require("os").hostname();
        console.log(" the primary server endpoint url is ", endpointUrl );

    });
}
server.initialize(post_initialize);

```

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : catalina 10.15
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  : 2.5.0-alpha.10

* **Node:** 
   `node --version` = v12.10.0

erossignon, 2020-02-27:
RefreshFunc callback expects a DataValue and not a Variant

replace 
```javascript 
namespace.addVariable({
            componentOf: device,
            nodeId: "ns=1;s=iota-mam-writer-reader",
            browseName: "mam reader",
            dataType: "String",
            value: {
                refreshFunc: async function (callback) {
                    setTimeout(()=>{
                        callback(
                            null, 
                            new opcua.Variant({dataType: opcua.DataType.String, value: "reply after 1s" })
                        )
                    }, 1000)
                },
```
with 
```javascript 
namespace.addVariable({
            componentOf: device,
            nodeId: "ns=1;s=iota-mam-writer-reader",
            browseName: "mam reader",
            dataType: "String",
 value: {
                refreshFunc: function (callback) {
                    setTimeout(()=>{
                        callback(
                            null, 
                            new opcua.DataValue({
                                statusCode: opcua.StatusCodes.Good,
                                value:  new opcua.Variant({
                                    dataType: opcua.DataType.String, 
                                    value: "reply after 1s" 
                                })
                            }) 
                        );
                    }, 1000)
                },
```

mtrabelsi, 2020-02-28:
thanks! worked! 

BTW is there any plan to make this SDK certified with OPC UA Foundation?

thanks

erossignon, 2020-02-28:
Marwen , the node-opcua sponsoring program will help us to grant the certification process, https://github.com/sponsors/node-opcua , I encourage you to consider it.

closed 2020-02-28 by erossignon