node-opcua/node-opcua #738: Reading  Datablock variables

mahboobelahi93, 2020-03-05:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com) (contact (@) sterfive.com )

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` =

erossignon, 2020-03-06:
please specify : node-opcua version: :

mahboobelahi93, 2020-03-06:
Hi,
I am using "node-opcua" 2.4.4 version and 2.4.4 version of "node-opcua-client".

erossignon, 2020-03-06:
Please try with node-opcua@2.5.0-alpha.14 and it should work

mahboobelahi93, 2020-03-06:
Okay,thanks for your support, also can you send me the documentation link of new version because I only see the documentation of previous version @ https://www.npmjs.com/package/node-opcua-client

erossignon, 2020-03-06:
Yes sure; you can get  here as a set of comprehensive and documented examples : see https://leanpub.com/node-opcuabyexample

teropes, 2020-06-27:
Hello,
I'm also working with Siemens PLC's with node-opcua. Here are some of my nodeId's as const string which do work:
`const NODE_ID_SAFETY = 'ns=3;s="generalStatus"."safety"';`

Notice how the complete string is wrapped inside single quotes to support it having double quotes in the nodeId itself. I had some struggle with these as well when I started. You can copy the exact nodeId from UAExpert right side attributes menu "copy value" -> wrap it inside single quotes since the string has double quotes -> it works.

I also recommend you don't rely on the fixed ns=3 in your application code, instead resolve the siemens namespace when you initialize your opc ua client and use it from there in all following reads, writes, subscriptions etc.

My connect function handles it this way:
```
  /**
   *  Connect to OPC UA server.
   */
  async connect() {
    const self = this;

    try {
      // 1. connect to server
      await self.opcClient.connect(self.config.endpoint);
      self.logger.log("info", `[opcua] Connected to server ${self.config.endpoint}.`);

      // 2. create session
      self.opcSession = await self.opcClient.createSession();
      self.logger.log("info", "[opcua] Session created.");

      // 3. init data subscription
      await self.initSubscription();
      self.logger.log("info", "[opcua] Subscription initialized.");

      // 4. read namespace array and set application namespace index
      if(self.config.applicationNamespace) {
        // this is required to read before we can resolve namespace indexes using getNamespaceIndex
        self.namespaceArray = await self.opcSession.readNamespaceArray();

        // get numeric value of application namespace, will be used to prefix nodeId's with it if ns= is not passed to nodeId
        // applications should not rely on fixed numeric ns= values, instead resolve it by namespaceURI when connecting
        self.ns = await self.opcSession.getNamespaceIndex(self.config.applicationNamespace);
        self.logger.log("info", `[opcua] resolved app ns ${self.config.applicationNamespace} to ns=${self.ns}`);

        if(self.config.cacheDataTypes) {
          self.logger.log("info", `[opcua] caching custom datatypes ${self.config.cacheDataTypes}`);
          for(let i=0; i < self.config.cacheDataTypes.length; i++) {
            await self.createObject(self.config.cacheDataTypes[i]);
          }
          self.logger.log("info", `[opcua] cached custom datatypes.`);
        }
      }

    } catch (error) {
      self.logger.log("error", "[opcua] connect() throws exception.");
      throw error;
    }
  }

```

My OPC UA client is javascript class which i create instance of in my application. I pass it config json object which includes some basic connection properties and also the applicationNamespace property which in case of Siemens S7 OPC UA servers has been `"http://www.siemens.com/simatic-s7-opcua"`. It resolves the numeric namespace id of it once and I then use it in all further reads, writes etc. I configure my nodeIds to only have to s=<stringNodeId> or i=<numericNodeId> part and my OPC UA client handles prefixing it with the application namespace index in case i don't provide it. Here is example read function from my class:
```
  /**
   *  Read node data.
   */
  async read(nodeId, indexRange) {
    const self = this;

    if(!self.opcSession) throw Error("OPC UA Session not available for reading data.");

    let range = indexRange ? indexRange : null;

    // read node value
    try {
      const completeNodeId = (!nodeId.includes("ns=")) ? `ns=${self.ns};${nodeId}` : nodeId ;
      const dataValue = await self.opcSession.read({ nodeId: completeNodeId, attributeId: opcua.AttributeIds.Value, indexRange: range });
      // return the actual payload as json object
      return dataValue.value.toJSON().value;

    } catch (error) {
      throw error;
    }
  }
```

closed 2021-05-08 by erossignon