node-opcua/node-opcua #739: Don't work session.createSubscription2

mariodiciccio, 2020-03-06:
I have node -v == 10.9.0 and node-opcua -v == 0.5.5

Why my code don't works?

var opcua = require('node-opcua')

var endpointUrl = 'opc.tcp://aasci.local:4334/UA/MyLittleServer'
var client = new opcua.OPCUAClient({
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 10,
    maxDelay: 200
  },
  keepSessionAlive: true
})

// console.log(client);

;(async () => {
  try {
  	await client.connect(endpointUrl)
  	let session = await client.createSession()

  	let datavalues = await session.read([{ nodeId: 'ns=1;b=1020FFAA' }]) // works as expected
  	console.log(datavalues);
  	
  	/*let subscription = opcua.ClientSubscription.create(session, {
        requestedPublishingInterval: 1000,
        requestedLifetimeCount:      100,
        requestedMaxKeepAliveCount:   10,
        maxNotificationsPerPublish:  100,
        publishingEnabled: true,
        priority: 10
    });*/
    
  	/*let subscription =  new opcua.ClientSubscription(session,{
                requestedPublishingInterval: 500,
                requestedLifetimeCount: 10,
                requestedMaxKeepAliveCount: 5,
                maxNotificationsPerPublish: 10,
                publishingEnabled: true,
                priority: 1
            });*/
            
        const subscription = await session.createSubscription2({
            requestedPublishingInterval: 2000,
            requestedMaxKeepAliveCount: 20,
            requestedLifetimeCount: 6000,
            maxNotificationsPerPublish: 1000,
            publishingEnabled: true,
            priority: 10
        });
	
	await session.close();
  	await client.disconnect();
  } catch(err) {
   console.log('Error: ', err.message);
  }
  
})()

result

Error
    at assert (/Users/aasci/Desktop/opcua/node_modules/node-opcua-assert/dist/index.js:6:21)
    at ClientSession.createSubscription2 (/Users/aasci/Desktop/opcua/node_modules/node-opcua-client-proxy/src/proxy.js:834:5)
    at /Users/aasci/Desktop/opcua/opcua-client.js:41:44
    at process._tickCallback (internal/process/next_tick.js:68:7)

erossignon, 2020-03-07:
please use node-opcua@2.4.0  , older versions of node-opcua do not support async/await syntax.

mariodiciccio, 2020-03-08:
thanks for reply
but also this script don't works
<pre>
var opcua = require('node-opcua')

var endpointUrl = 'opc.tcp://127.0.0.1:4334/UA/MyLittleServer'
var client = new opcua.OPCUAClient({
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 10,
    maxDelay: 200
  },
  keepSessionAlive: true
})

client.connect(endpointUrl, (err) => {
  console.log("CONNECTED TO PLC");
  if (err){
    console.error("PLC CONNECT ERROR", err);
    return;
  }

  getSession();
});

function getSession() {
  client.createSession( function(err, session) {
     if(err) {
         console.error('Session Error');
         return;
     }

     session.readVariableValue('ns=1;b=1020FFAA', function(err, dataValue) {
       console.log(dataValue);
   
     });

     let subscription =  new opcua.ClientSubscription(session,{
                 requestedPublishingInterval: 500,
                 requestedLifetimeCount: 10,
                 requestedMaxKeepAliveCount: 5,
                 maxNotificationsPerPublish: 10,
                 publishingEnabled: true,
                 priority: 1
    });

    subscription.on("started",()=>console.info("opcua session started"));
    let monitoredItems = [];

    const itemToMinitor = {
         nodeId: opcua.resolveNodeId('ns=1;b=1020FFAA'),
         attributeId: opcua.AttributeIds.Value
    };

    const monitoringParameters= {
        samplingInterval: 10,
        discardOldest:    true,
        queueSize:        1000
    };

    const monitoredItem = client.ClientMonitoredItem.create( subscription, itemToMinitor, monitoringParameters, client.TimestampsToReturn.Both);
    monitoredItem.on("changed",function(dataValue) {
        console.log("dataValue", dataValue.toString());
    });

   });
}

</pre>

Result 
<pre>
CONNECTED TO PLC
ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property 'create' of undefined
    at /Users/user/Desktop/opcua2/opcua-client.js:63:54
    at /Users/user/Desktop/opcua2/node_modules/node-opcua-client/src/opcua_client.js:655:17
    at /Users/user/Desktop/opcua2/node_modules/node-opcua-client/src/opcua_client.js:545:24
    at /Users/user/Desktop/opcua2/node_modules/node-opcua-client/src/client_session.js:1178:9
    at modified_callback (/Users/user/Desktop/opcua2/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1010:28)
    at process_request_callback (/Users/user/Desktop/opcua2/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:78:5)
    at ClientSecureChannelLayer._on_message_received (/Users/user/Desktop/opcua2/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:129:5)
    at MessageBuilder.emit (events.js:182:13)
    at MessageBuilder._decode_message_body (/Users/user/Desktop/opcua2/node_modules/node-opcua-secure-channel/src/message_builder.js:462:22)
    at MessageBuilderBase._feed_messageChunk (/Users/user/Desktop/opcua2/node_modules/node-opcua-transport/src/message_builder_base.js:196:18)
Session Error
DataValue {
  value:
   Variant {
     dataType: { [Number: 11] key: 'Double', value: 11 },
     arrayType: { [Number: 0] key: 'Scalar', value: 0 },
     value: 16,
     dimensions: null },
  statusCode:
   { [Number: 0] value: 0, description: 'No Error', name: 'Good' },
  sourceTimestamp: null,
  sourcePicoseconds: 0,
  serverTimestamp: null,
  serverPicoseconds: 0 }
opcua session started

</pre>

erossignon, 2020-03-08:
If analyze the error:

```` ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property 'create' of undefined
    at /Users/user/Desktop/opcua2/opcua-client.js:63:54
```` 

line 63 in your code (opcua-client.js) is:
````javascript
const monitoredItem = client.ClientMonitoredItem.create( subscription, itemToMinitor, monitoringParameters, client.TimestampsToReturn.Both);   
````

I guess you need to replace `client.ClientMonitoredItem.create` with `opcua.ClientMonitoredItem.create`.

you could find more comprehensive in  https://leanpub.com/node-opcuabyexample

mariodiciccio, 2020-03-09:
I am sorry, but don't understand:

<pre>
monitoredItem = opcua.ClientMonitoredItem.create( subscription, itemToMinitor, monitoringParameters, opcua.TimestampsToReturn.Both);
</pre>

I will buy the book soon

mariodiciccio, 2020-03-17:
I resolved using node-opcua 2.4.4 with the plc siemens firmware 2.8.1.
All works fine.

closed 2020-04-27 by erossignon