node-opcua/node-opcua #740: Asynchronous set with response based on set status

cdh4u, 2020-03-08:
**Is your feature request related to a problem? Please describe.**
I read some of the old discussions (#35) regarding server-side asynchronous set and being able to wait for the set result before returning the response coded. I have been trying to look for information about that, but I haven't found it, so I wonder what the current status is. For me it would be essential for the server to wait for the set status before it returns a response code. 

I am new to node-opcua, so my appologies is there is something I have missed. Thank You!

**Describe the solution you'd like**
An asynchronous set function that allows me to return a status code based on the set result.

**Describe alternatives you've considered**
Using the current function with some loop waiting for the set result before sending the response code.

erossignon, 2020-03-08:
This is covered in the book ["node-opcua by example"](https://leanpub.com/node-opcuabyexample) 5.5.7 technique 6 - using a asynchronous timestamped getter and setter

cdh4u, 2020-03-08:
> This is covered in the book "node-opcua by example" 5.5.7 technique 6 - using a asynchronous timestamped getter and setter

Thanks! I have purchased the book.

closed 2020-03-08 by cdh4u