node-opcua/node-opcua #741: Create ObjectTypes with Variables

cdh4u, 2020-03-10:
**Is your feature request related to a problem? Please describe.**

Is there some documentation and/or examples on how to define your own ObjectTypes/VariableTypes, and use those for Objects/Variables?

I have written the following code:

                //Define ObjectType
		objectDeviceType = namespace.addObjectType({
			browseName: "objectDeviceType"
		});

                //Define variable for myObjectType
		namespace.addVariable({
			propertyOf: objectDeviceType,
			browseName: "myVariable",
			dataType: "Int16"
		});
  
                //Note that I don't define the getter and setter functions, because they would be 
                //difference in each Object of the ObjectType. Is that even possible?

		objectDevice1 = namespace.addObject({
			componentOf: objectFolder,
			browseName: "objectDevice1",
			typeDefinition: objectDeviceType
		});

When I run the server code, and use the Prosys client to connect to the server:

- I can see myObjectType in the Types folder
- I can see  objectDevice1 in the Objects folder, with HasTypeDefinition = objectDeviceType
- I can NOT see a HasProperty reference to myVariable within objectDevice1

**Describe alternatives you've considered**
Define each Object of the same ObjectType as separate Objects (with the default HasTypeDefinition = BaseObjectType)

erossignon, 2020-03-22:
have a look at the "boiler" test in the code

cdh4u, 2020-03-23:
A question is regarding the follwoing code in boiler_system.ts:

    // --------------------------------------------------------
    // GenericSensorType  <---- GenericControllerType
    // --------------------------------------------------------
    const genericControllerType = namespace.addObjectType({
        browseName: "GenericControllerType"
    });

Should there be 'subtypeOf:genericSensorType'?

I.e.:

    // --------------------------------------------------------
    // GenericSensorType  <---- GenericControllerType
    // --------------------------------------------------------
    const genericControllerType = namespace.addObjectType({
        browseName: "GenericControllerType",
        subtypeOf:genericSensorType
    });

erossignon, 2020-10-02:
* The model exhibits two different concepts : Sensors and Controllers. Controllers does not derive from Sensors and vice-versa. 

* install_extra_properties extends the UAObject with javascript properties that match the UA Components and Properties of the object, so that instead of writing;
```` 
const output =   genericSensorType.getComponentByName("Output");
console.log(output.nodeId.toString());
```` 
you could write
```` 
const output =   genericSensorType.output;
console.log(output.nodeId.toString());
```` 

* instantiate is needed whenever you want to create an instance of  given ObjectType or VariableType.
As you said, this is done most commonly to create the instance that will be store in the "Objects" folder. But it could be also useful when defining types that contains instances of other Variable or Object types.

closed 2020-10-02 by erossignon