node-opcua/node-opcua #742: openssl-1.0.2o-x64_86-win64.zip error 404

belqit, 2020-03-11:
[image]

https://github.com/node-opcua/node-opcua-pki/files/1886617/openssl-1.0.2o-x64_86-win64.zip 
This url is not working anymore, there is a way to fix this problem without update the module?

michaelgrafl, 2020-03-19:
@belqit You can copy a local openssl.exe into the folder where it looks for it. It will need the appropriate DLLs, too, of course (there are two on my system).

Maybe it would make sense to host the openssl binaries on some place that the package maintainers have control over. I'm having the same problem with another version and another URL.

erossignon, 2020-03-19:
Hi @belqit , could you fill the template form ? We do need the  version of node-opcua you are trying to use,

Recent version of node-opcua-pki refers to https://github.com/node-opcua/node-opcua-pki/releases/download/v1.1.0/openssl-1.0.2r-x64_86-win64.zip instead 

Please upgrade.

closed 2020-03-27 by stale[bot]