node-opcua/node-opcua #745: Please make the location of the certs folder 'pki' configurable.

Disco1267, 2020-03-12:
The folder 'pki' at the project root is the place where the opcua client looks for trusted certs and also drops the rejected ones.
I would like to have a property or Env var that we could use to determine the location and/or name of this folder.
The purpose is to get the pki folder out of the node project folder and instead have it in a place where an administrative user of the solution would look for it, in order to sort out rejected and accepted certs.
Yes, the certificate file and private key file location can be passed in as properties to  OPCUAClient.create via OPCUAClientOptions structure, but rejected and trusted ones always end up in the default location.

Perhaps this is already possible and I just wasn't able to find it, then please add a line in the README.md about it.

Thank you

mikakaraila, 2020-03-12:
This way:
const certificateManager = new opcua.OPCUACertificateManager({
  rootFolder: path.join(path.dirname(process.execPath), "certificates/PKI"),
  automaticallyAcceptUnknownCertificate: true
});

Disco1267, 2020-03-13:
Yes, in the meantime I had discovered this Class.
But I was wondering if that instance would become automatically a global singleton instance.
...
Don't mind I will try that now.

Disco1267, 2020-03-13:
Sorry, this didn't work. It seems the OPCUACertificateManager needs to be injected somewhere somehow.
I should mention that I am working on a OPCUA client. Searching through the node packages I found that OPCUAServerOptions and OPCUAServerEndPoint seem to have properties for different CertManagers, but the client ?

maanenh, 2020-03-13:
We are working on the same issue. In our environment we run the opcua client inside a Lambda on AWS Greengrass. In that environment there is no permission to create files or folders in the working directory.
And in this function there is a TODO comment:  https://github.com/node-opcua/node-opcua/blob/42564da47ba4b97a4949b31e94ee745d90cfa303/packages/node-opcua-client/source/private/client_base_impl.ts#L178

Disco1267, 2020-03-13:
OK, it doesn't even use a CertManager object. And strangely that function despite of its name doesn't even verify anything, i just dumps it into  
`const pkiFolder = process.cwd() + "/pki";`
Which means the location of the folder depends on the current dir when the client was started. This is what I just found out by experimenting. 
It also explains why certs which have already moved to 'trusted' keep re-appearing in 'rejected'.

mikakaraila, 2020-03-13:
You need to use it through server_options when you create your server.

const server_options: opcua.OPCUAServerOptions = {
  securityPolicies: policies,
  securityModes: modes,
  certificateFile: server_certificate_file,
  privateKeyFile: server_certificate_privatekey_file,
  port,
  allowAnonymous: true,
  serverCertificateManager: certificateManager, // Use here!
  ....
}

Then pass it to constructor:
const server = new OPCUAServer(server_options);

mikakaraila, 2020-03-13:
With client similar way:
 const options2: opcua.OPCUAClientOptions = {
    clientName,
    securityMode,
    securityPolicy,
    serverCertificate,
    defaultSecureTokenLifetime: 60000 * 5, // 5 min
    endpoint_must_exist: false,
    keepSessionAlive: true,
    requestedSessionTimeout: 60000, // 60 sec
    connectionStrategy: {
      maxRetry: 3, // 10512000,      // 10 years should be enough. No infinite parameter for backoff.
      initialDelay: 3000,
      maxDelay: 30 * 1000
    },
    // If not from server endpoint use local certificates
    certificateFile: path.join(path.dirname(process.execPath), "/certificates/Client_selfSigned.pem"),
    privateKeyFile: path.join(path.dirname(process.execPath), "/certificates/PKI/own/private/private_key.pem")
  };

And then pass options when you create client:
client = opcua.OPCUAClient.create(options2);

Disco1267, 2020-03-13:
@mikakaraila wrote
> You need to use it through server_options when you create your server.

But the issue is about the client. As pointed out by @maanenh the client doesn't use the CertManager at all. And nothing can keep it from dropping _every_ server certificate into $CWK/pki/rejected.

And yes, we can pass any path to our clients certificate file, but the location of trusted and/or rejected server certs is out of control. Even more the client accepts every certificate. Validation is just not implemented.

At least I found  a workaround for the validation:
Create an Instance of the OPCUACertificateManager as shown above.
Connect e.g.
`
const client = OPCUAClient.create(options);
//now validate yourself:
const certificateManager = new OPCUACertificateManager(...);
const serverTrustState = await certificateManager.isCertificateTrusted(client.serverCertificate);    assert.strictEqual(serverTrustState, "Good", 'Server Certificate is:' + serverTrustState);
`

maanenh, 2020-03-13:
I think this does not solve the issue in the client that is in:  
https://github.com/node-opcua/node-opcua/blob/42564da47ba4b97a4949b31e94ee745d90cfa303/packages/node-opcua-client/source/private/client_base_impl.ts#L178

Disco1267, 2020-03-13:
Yes, you could only work it around in an clumsy way by starting up like this:
```sh
cd /tmp
node /my/project/folder
```
This way _pki_ would end up it tmp.
Or alternatively patch the code by simply emptying  
function _verify_serverCertificate
and implement the validation like I have pointed out above.

gunar393, 2020-03-13:
> The folder 'pki' at the project root is the place where the opcua client looks for trusted certs and also drops the rejected ones.
> I would like to have a property or Env var that we could use to [...]

Hi, there is no trusted folder in the pki. There exists only rejected folder where the untrusted certificate is thrown. Are you able to trust the rejected certificate?

gunar393, 2020-03-13:
> This way:
> const certificateManager = new opcua.OPCUACertificateManager({
> rootFolder: path.join(path.dirname(process.execPath), "certificates/PKI"),
> automaticallyAcceptUnknownCertificate: true
> });

can you please where this section of code exists

maanenh, 2020-03-13:
We are running our code inside a Lambda on AWS Greengrass. In that environment we are not able to change working directory. 
For now we are going to patch function _verify_serverCertificate

maanenh, 2020-03-13:
@gunar393 See:  https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-certificate-manager/source/certificate_manager.ts

mikakaraila, 2020-03-13:
For AWS Greengrass you should add local resources and then use environment variables to make it clean.

maanenh, 2020-03-13:
The rejected folder during runtime is: /lambda/pki/rejected.
I created a local recource with read/write access to that folder and mapped it to a folder on the greengrass host. This resulted in the next error.
```
runtime execution error: unable to start lambda container.    {"errorString": "failed to run container sandbox: container_linux.go:344: starting container process caused \"process_linux.go:424: container init caused \\\"rootfs_linux.go:64: mounting \\\\\\\"/usr/rejected\\\\\\\" to rootfs \\\\\\\"/greengrass/ggc/packages/1.10.0/rootfs/merged\\\\\\\" at \\\\\\\"/lambda/pki/rejected\\\\\\\" caused \\\\\\\"lstat /greengrass/ggc/packages/1.10.0/rootfs/merged/lambda/pki: permission denied\\\\\\\"\\\"\""}
```
For now we are going to patch function _verify_serverCertificate

Disco1267, 2020-03-13:
@gunar393 wrote
> Hi, there is no trusted folder in the pki. There exists only rejected folder where the untrusted certificate is thrown. Are you able to trust the rejected certificate?
The current client trusts _every_ certificate. It doesn't even look at the trusted folder

> can you please where this section of code exists
I wrote those lines. It demonstrates how you can do the missing validation yourself.
After connecting you can take the servers cert. 
So we create an instance of the CertificateManager and then use it for validation of the server certificate..
The certificate manager respects the trusted folder certs.

mikakaraila, 2020-03-13:
Greengrass runs lambda functions in sandbox, you have to add resource

mikakaraila, 2020-03-16:
You can turn off this:
automaticallyAcceptUnknownCertificate: true => false

DG1000c, 2020-04-07:
I am using this is as part of a nodered docker container and I am faced by this issue too. The container stops and cannot be started properly again, cause it complains that there is no permission for `mkdir /usr/src/node-red/pki` and the thrown exceptions makes the container stop.

Since nodered docker container stores all user data in directory `/data` of the container which is typically mapped to a host directory or to a virtual volume, these data would have been lost on container recreation.

I manually patched the code already mentioned above from

`const pkiFolder = process.cwd() + "/pki";`

to

`const pkiFolder = "/data/pki";`

which then works.

So my requirements are:

1. Avoid stop of docker container when the directory for `pki` cannot be created
2. Make this directory configurable via nodered config

erossignon, 2020-05-30:
This directory is already configurable inside node-opcua.

const certificateManager = new OPCUACertificateManager({
    rootFolder : '/home/me/pki', // put the name of a folder that you have RW access to
     automaticallyAcceptUnknownCertificate: false
});
// in order to create the pki folder and subfolders, please call initialize()
await certificateManager.initialize();
````

erossignon, 2021-01-17:
node-opcua@2.29 further improve client dealing with certification verficiation. client  now uses a Certificate Manager and perform exhaustif certificate checks.

closed 2021-01-17 by erossignon