node-opcua/node-opcua #760: Statuscode always "Good" even when no initial value comes

aki-0919, 2020-04-15:
**Current behavior**

In the OPC UA client the Statuscode shows always "Good" even when OPC UA server receives no value after it starts up.

**Describe the bug**
According to OPC UA specification(Spec 1.04 Part 4 section 7.34.2 Table 178 last entry), the Stasucode should show "Bad_WaitingForInitialData" when waiting for the server to obtain first value .
I'm currently trying to update the package version from 0.7 to 2.5.4.
It is working as I expected with 0.7.0 but not with 2.5.4.
It seems this [line](https://github.com/node-opcua/node-opcua/blob/v2.5.4/packages/node-opcua-address-space/src/ua_variable.ts#L589) set the statuscode "Good" as default.

**To Reproduce**
Steps to reproduce the behavior:
1. .... Start any server
2. ...... With a client browse any node on which no data is received after server startup
3....... Observe Statuscode field

**Expected behavior**
Stasucode should show "Bad_WaitingForInitialData" when OPC UA server receives no value after it starts up.

**Screenshots**
- node-opcua version:2.5.4
[image]

- node-opcua version:0.7.0
[image]

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
   (*)  my request is related to node-opcua acting as a OPCUA SERVER 

   (*) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
   (*) I am using an application that uses node-opcua 
  
       - (*) node-red
       - ( ) other : please specify 
        
  * Device: IPC
  * OS version: _________
    
    - (*) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :2.5.4

* **Node:** 
   `node --version` =10.19

aki-0919, 2020-04-15:
Is this procedure going as expected?
I'm not sure if I understand it correctoly but, `bindVariable()`([here](https://github.com/node-opcua/node-opcua/blob/v2.5.4/packages/node-opcua-address-space/source/loader/load_nodeset2.ts#L1218)) sets values for each node if server receives them. And Statuscode is set as "Good" in `setValueFromSource()`

However, in this [line](https://github.com/node-opcua/node-opcua/blob/v2.5.4/packages/node-opcua-address-space/src/ua_variable.ts#L296), default value is set in `makeDefaultVariant()` even for unreceived value. This leads that `if (options.value){ this.bindVariable(options.value); }` is always true.
Therefore, on creating node process, `bindVariable()` is called for every node and the Statuscode of them are set "Good" whichever the server receives value or not.

erossignon, 2020-05-06:
Yes, this behavior is expected especially from values coming from external nodeset.
and the change was intentional.

closed 2021-05-08 by erossignon