node-opcua/node-opcua #762: OPCUA client connection timeout

pminel, 2020-04-16:
**I'm submitting a ...**
 - (X) bug report
 - ( ) feature request
 - ( ) support request

**Current behavior**
SERVER IS TURNED OFF TO TEST CONNECTION ERROR
Opcua client try to connect only for 1 time but i receive the "connection_failed" event after about 1 minute.

**Expected behavior**
I expect that client start the connection immediately (initialDelay=0) and for 1 attempt (maxRetry=0).

**Minimal reproduction of the problem with instructions**
Instruction: 
1. client = opcua.OPCUAClient.create({ keepSessionAlive: true, connectionStrategy: { initialDelay: 0, maxRetry: 0 } })
2. client.on('connection_failed', (err) => { logger.log('Connection failed at ' + moment().format('HH:mm:ss.SSS')) })
3. client.connect(endpoint, err => { if(err) { logger.log('Not connected at ' + moment().format('HH:mm:ss.SSS')) } })

Output:
Trying connect at 14:18:34.483 // connection timeout about 1 minute
Connection failed at 14:19:36.500
Not connected at 14:19:36.503

  
**What is the motivation / use case for changing the behavior?**
I would return connection status as faster as possible to client without waiting about 1 minute

**Please tell us about your environment:**

  * (X) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (X) Linux   : version : 18.04
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: LMC
     - Version: Pro2
     - Manufacturer/Software vendor: Schneider Electric
     - link : https://
     
    
* **node-opcua version:**
2.5.7

* **Node:** 
node --version=v10.16.3
npm ..version=6.13.7

erossignon, 2020-04-26:
try 
````
const client = OPCUAClient.create({ 
  connectionStrategy: { 
         initialDelay: 100,
         maxRetry: 1
 });
````

pminel, 2020-04-27:
Good!
With this configuration i've got connection response in 5/6 seconds...
Thanks

closed 2020-04-27 by erossignon

clockshock, 2024-04-06:
I'm encountering the same issue. Regardless of the values I use for 'initialDelay' or 'maxRetry', it takes at least 20 seconds before I receive a connection error.

With this takes about 40s to get the connection error:
`  const connectionStrategy: ConnectionStrategyOptions = {
    initialDelay: 100,
    maxRetry: 1,
  };`

With this takes about 20s to get the connection error:
`  const connectionStrategy: ConnectionStrategyOptions = {
    initialDelay: 100,
    maxRetry: 0,
  };`

Is there a way to decrease the timeout?

My current node-opcua version is: "^2.123.0"

AndreasHeine, 2024-04-06:
@clockshock maybe have a look at this:  https://github.com/node-opcua/node-opcua/blob/master/packages%2Fnode-opcua-client%2Fsource%2Fclient_base.ts#L139

i guess its a net.socket topic 🤔

clockshock, 2024-04-07:
@AndreasHeine  yes if i set `transportTimeout: 5000` i get this error message after about 5 seconds:

`The connection may have been rejected by server,
Err = (premature socket termination socket timeout : timeout=5000  ClientTCP_transport12)`

 Thank you Andreas.

AndreasHeine, 2024-04-07:
> @AndreasHeine  yes if i set `transportTimeout: 5000` i get this error message after about 5 seconds:
> 
> `The connection may have been rejected by server,
> Err = (premature socket termination socket timeout : timeout=5000  ClientTCP_transport12)`
> 
>  Thank you Andreas. 

make sure the timeout is not to small so the os can free up ressources anf you dont run into exauthion!