node-opcua/node-opcua #765: Versioning in package.json makes it impossible to choose a node-opcua version older than most current one

cstim, 2020-04-22:
**Current behavior**
The various `package.json` files throughout the sub-packages refer to each other by specifying the version **with prefixing caret**, like so: `"^2.4.4"`. However, this means at any build time in the future, they won't pull in the given version number but instead any most up-to-date version number with the same major number `2.x.x`. The problem is that as a library user, I may have reasons that I want to stick to a specific version of node-opcua, for example because of newer dependencies that are pulled in by newer versions of node-opcua which I don't want. But because your package.json version dependencies are written with the caret prefix, even when I chose `node-opcua: "2.4.2"` without caret in *my* package.json, I won't get node-opcua in that version but instead in the most recent 2.x.x version. Well, to be totally exact I do get the top-level package in that version, but the actual code in all your sub-packages will not be the version I asked for but will be some future version that I didn't know about when I wrote my implementation against some specific version of node-opcua.

The particular occasion where I discovered this is that in `2.5.8`, your code introduced the requirement of `chalk:4.0.0` which in turn requires `node>=10` but my project was using `node=8`, so my project no longer builds and I am unable to specify that I want some node-opcua version earlier than 2.5.8.

**To Reproduce**
Steps to reproduce the behavior:
1. Write a user project with package.json that asks for any node-opcua version older than the most recent one, say `node-opcua: "2.5.7"`
2. Run npm install
3. Discover that the most recent node-opcua-common and all other node-opcua-* packages are pulled in, say `node-opcua-common: 2.5.8` instead of the older version I asked for

**Expected behavior**
Your cross-referencing of your subpackages in your `package.json`  should be specified only with exact version numbers, like so: `node-opcua-data-model: "2.5.1"` instead of the caret prefix. Only this way a user will actually be able to decide whether they want npm's automatic version upgrade, in which case they will specify 'node-opcua: "^2.5.1"` with caret in their package.json, or whether they don't want it, in which case they specify this without the caret. 

* **node-opcua version:**  : 2.5.8 and all earlier, all the way back since 0.x times

erossignon, 2020-04-28:
^ means that minor version update are possible.

it is normal that "2.5.7" goes to "2.5.9" because  this is a patch update.

see https://docs.npmjs.com/about-semantic-versioning

version 2.5.9 onward will used fixed patch update only ( without ^) instead of "minor update" (^)

erossignon, 2020-05-19:
In fact, it seems that lerna, the tool used to publish the node-opcua-* packages rewrites all version as "^x.y.z" (minor update) and defeat the strategy of forcing patch update only.

erossignon, 2020-05-19:
@cstim 
>>The particular occasion where I discovered this is that in 2.5.8, your code introduced the requirement of chalk:4.0.0 which in turn requires node>=10 but my project was using node=8, so my project no longer builds and I am unable to specify that I want some node-opcua version earlier than 2.5.8.

In fact, chalk:4.0.0 is working well with nodejs-8 , it should not be a problem for you.

Please check node-opcua compiling and testing well on travis with node-opcua 8 for instance https://travis-ci.org/github/node-opcua/node-opcua/jobs/684953580

cstim, 2020-05-20:
> In fact, chalk:4.0.0 is working well with nodejs-8 , it should not be a problem for you.

I'm confused. In chalk:4.0.0, its `package.json` contains the entry `"engines": { "node": ">=10" }` and I thought this means chalk requires node => 10 as engine, which matches our experience that it won't work with node 8. What am I missing here?

cstim, 2020-05-20:
> In fact, it seems that lerna, the tool used to publish the node-opcua-* packages rewrites all version as "^x.y.z" (minor update) and defeat the strategy of forcing patch update only.

Yes, I can observe that, too. That's unfortunate. Thanks for your efforts, though!

erossignon, 2020-05-20:
@cstim 
chalk:4.0.0,  it appears that n chalk:4.0.0,  is working well on nodejs V8 as proved by the tests.
`"engines": { "node": ">=10" }` is just an indication in package.json and not an enforced rule.

If you are using typescript you may have to revisite the way you import chalk 
from 
````
import chalk from "chalk";
````
to 
````
import * as chalk from "chalk";
````

May be that's it. your post does not clearly describe the type of error you are experiencing with chalk.

erossignon, 2020-08-11:
@cstim  
Good news... 
adding  `"exact": true` to the  `lerna.json` file prevents lerna to replace "x.y.z" with "^x.y.z" when updated mono-repo modles version number during version increase.
I guess we now have a scheme in place to nail down node-opcua to a older version

There are some implication though
  -  this  will only work with version 2.10.0 onward and will not be working with old versions already released.
  -  user should only import "node-opcua" and not sub modules (node-opcua-xxx) to prevent accidental mistmatch. This should not be a problem as "node-opcua" modules exports the entire public API.

closed 2020-09-23 by erossignon