node-opcua/node-opcua #769: BadNodeIdUnknow - adding new Nodes

NoelLD, 2020-04-28:
I have a problem adding a new node to the server. The aim is that a client is able to add nodes in servers object model. Every Clients gets own "root node" where nodes can be add, remove and eddit by "owner client". I have written some functions for this purpose but I'm stucked.
Here the Client should add an own root Node:
(until know, function is called from an a testing forms)
```
public BaseObjectState addClientRootNode(string name, BaseObjectState parentNode= null)
        {
            FolderState objfolder = new FolderState(parentNode);

            objfolder.SymbolicName = name;
            objfolder.ReferenceTypeId = ReferenceTypes.Organizes;
            objfolder.TypeDefinitionId = ObjectTypeIds.FolderType;
            objfolder.NodeId = new NodeId(2003, 3); // identifier 2003 just for testing, unused in model
            objfolder.BrowseName = new QualifiedName(name, NamespaceIndex);
            objfolder.DisplayName = new LocalizedText(name);
            objfolder.WriteMask = AttributeWriteMask.None;
            objfolder.UserWriteMask = AttributeWriteMask.None;
            objfolder.EventNotifier = EventNotifiers.None;
            if (parentNode != null)
            {
                parentNode.AddChild(objfolder);
            }
            return null;
        }
```

I used UAExpert as a client to see if it worked. The Node is added to server model, but i can't browse: BadNodeIdUnknow. It seems to add the node but not correctly.
[image]
[image]

detailed information:
.Net Framwork 4.6

By now, the methods is given the machines Node, which is defined in an xml. Xml is loading at the beginning the servers model. "machines" node is working correctly.

I'm confused because I almost copied the exmaple code.

erossignon, 2020-04-29:
Your code doesn't look like TypeScript but rather C# , are you posting to the right project ?

NoelLD, 2020-04-29:
oh yes I'm sorry

closed 2020-04-29 by NoelLD