node-opcua/node-opcua #771: GET UserName Identity on OPC UA Server when OPCUA Client connect?

tttranducdung, 2020-05-01:
When Clients create session with OPCUA Server, it will send userName & password Identity to activeSession on Server. How to I can know that "UserName" when client connect to Server on my Server (node-opcua)? Here is an example of Prosys OPC UA Simulation Server.
[image]

tttranducdung, 2020-05-11:
@erossignon can you help me to solve this problem?

erossignon, 2020-07-16:
you could capture it during the `"session_activate"` event:

````
    server.on("session_activated", (session) => {
        console.log("username is =", session.userIdentityToken.userName);
    });
```

closed 2020-11-30 by erossignon