node-opcua/node-opcua #772: Update mkdirp (security issue)

sanderd17, 2020-05-02:
**Describe your problem**
I got a security warning from github
The warning I got is the following.

CVE-2020-7598
> high severity
> Vulnerable versions: < 0.2.1
> Patched version: 0.2.1

> minimist before 1.2.2 could be tricked into adding or modifying properties of Object.prototype using a "constructor" or "proto" payload.

Tracing back the dependency chain, it appears the node-opcua-certificate-manager [depends on mkdirp@0.5.1](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-certificate-manager/package.json#L23). Which in turn depends on a vulnerable minimist version.

**Describe the solution you'd like**
mkdirp is meanwhile updated: 
* a version 1.0.x exists with a new API and no dependency on minimist
* a version 0.5.5 which depends on a newer [minimist@1.2.5](https://github.com/isaacs/node-mkdirp/blob/0.x/package.json#L22)

Updating to either version of mkdirp would fix the security warning.

**Additional context**
As far as I understand, this is mostly a security issue in a browser context. So it's of low importance for node-opcua. But it would still be nice to get rid of the warning.

erossignon, 2020-05-04:
Yes, right, mkdirp@1.0.0 introduced all sorts of nasty side effects on module we are relying upon (expecially lerna).  
This is why I temporaly decided to revert the changes.
We will definitely move to mkdir latest , but need to wait until it is fully integrated inside dependent module to avoid the pitfall.

erossignon, 2020-06-03:
mkdirp updated to 0.5.5 to fix issue with  vunlernability minimist . 
available on node-opcua@2.5.10 onward

closed 2020-06-03 by erossignon