node-opcua/node-opcua #783: node opcua client is not browsing nodes if the server is running in a CPU with very less memory.

Nandish-Reddy, 2020-05-08:
**Current behavior**

**Describe the bug**
I tried browsing opcua server using node opcua but it returns nothing on a browse request. The CPU where opcua server is running has very less memory like in few 150 kbs for code section and 1 mb for data section.

 
**To Reproduce**
Steps to reproduce the behavior:
1>Run a sample opcua server in a processor with 150kb for code and 1 mb for data section
2> see's no results, no browse data

**Expected behavior**
Browsed Data

 * (yes ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  

  * (yes ) I have installed node-opcua as a package ( using npm install )

  
    
        
  * Device: _____
  * OS version: _____
    
   - ( yes) Linux   : version : Debian 9

   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:OPCUA 
        - Manufacturer/Software vendor: Foundataion server
     
     

* **node-opcua version:**  :
node -opcua client verison 2.1.5

* **Node:** 
   `node --version` =

erossignon, 2020-05-08:
Hi strange,
Can you perform some diagnostics, it may be that the server is too slow and doesn't respond inside the timeout frame setup for the communication.
Can you enable the trace and provide a dump  ?  you have to setup the NODEOPCUADEBUG environement variable first. like this:
```` 
export NODEOPCUADEBUG=TRACE-REQUEST-RESPONSE
node myclient.js > dump.txt
````

Nandish-Reddy, 2020-09-01:
this issue is not reproducible.

closed 2020-09-01 by Nandish-Reddy