node-opcua/node-opcua #785: TypeScript Compile Error

seang96, 2020-05-18:
**Current behavior**

Building with TypeScript on a version > 2.5.10 after node-opcua-factory updated to v2.5.10.

**Describe the bug**
Fail to compile due to typescript files in node-opcua-factory failing my tsconfig checks.

```
node_modules/node-opcua-factory/source/factories_baseobject.ts:288:29 - error TS7018: Object literal's property 'lines' implicitly has an 'any[]' type.

288                             lines: [],
                                ~~~~~~~~~

node_modules/node-opcua-factory/source/factories_baseobject.ts:306:58 - error TS7018: Object literal's property 'lines' implicitly has an 'any[]' type.

306                 const data1 = { padding: padding + "  ", lines: [] };
```

See issue: https://github.com/microsoft/TypeScript/issues/15363

**To Reproduce**
Steps to reproduce the behavior:
1. Install node-opcua on latest version
2. Build with Typescript

**Expected behavior**
Build succeeds

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

 * (* ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * (* ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: Laptop
  * OS version:
    
    - (x) Windows : version : 10 (2004)
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:N/A
     - Version:N/A
     - Manufacturer/Software vendor:N/A
     - link : N/A
     

* **node-opcua version:**  :2.5.12

* **Node:** 
   `node --version` =v12.16.3

erossignon, 2020-05-18:
Yep, new type checked introduced by the most recent version of TypeScript 3.9.
Please use typescript 3.8 while we're fixing this in future version. this is wip

seang96, 2020-05-18:
@erossignon, I attempted to compile using 3.8 and 3.7.2 both of which still have the same error.

erossignon, 2020-05-19:
@seang96 , can you share your tsconfig ?
it might require some adjustement

closed 2020-05-19 by erossignon

erossignon, 2020-05-19:
related to https://github.com/node-opcua/node-opcua/pull/766

commit 60a65f2e (2020-05-19): fixes #785