node-opcua/node-opcua #788: Support for cluster mode

avalla, 2020-05-22:
Is there any way to support cluster mode efficiently?

e.g. using server listen in exclusive mode

erossignon, 2020-05-27:
you mean passing the `{ exclusive: true}` flag to `server.listen()` ?

avalla, 2020-05-28:
sorry, it was badly explained, as use { exclusive: false }

I added an opc-ua server in a clustered application and it was not working like express (same port multiple instances)

maybe it was bad architecture design, now my opc-ua server runs on a different (not clustered) process

erossignon, 2020-05-30:
@avalla , the architecture you choose makes sense, it is cleaner to have your node-opcua running inside its own process and separated from your express(like) app.

closed 2020-05-30 by erossignon