node-opcua/node-opcua #793: Unnecessary event emit - alarmChanged

aki-0919, 2020-05-29:
Hi,

In **if(!alarm) { ... }** block,  is <**this.emit("alarmChanged", alarm)**> really needed?
**alarm** is always **null** in this line. Maybe it could be removed.

[client_alarm_list.ts#L60](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/source/alarms_and_conditions/client_alarm_list.ts#L60)

    public update(eventField: EventStuff) {
        const { conditionId, eventType } = eventField;
        assert(conditionId, "must have a valid conditionId ( verify that event is a acknodweldgeable type");
        const alarm = this.findAlarm(conditionId.value, eventType.value);

        if (!alarm) {
            const key = this.makeKey(conditionId.value, eventType.value);
            const newAlarm = new ClientAlarm(eventField);
            this._map[key] = newAlarm;
            this.emit("newAlarm", newAlarm);
            this.emit("alarmChanged", alarm);   <------ THIS LINE
        } else {
            alarm.update(eventField);
            this.emit("alarmChanged", alarm);
        }
    }

erossignon, 2020-05-29:
Thank you , this need to be fixed , indeed..

maneetgoyal, 2021-01-10:
Considering using this library in a project; the issue seemed easy to fix so created a PR: https://github.com/node-opcua/node-opcua/pull/928. Hope that's okay.

closed 2022-04-16 by stale[bot]