node-opcua/node-opcua #796: ClientSession interface is missing ```deleteMonitoredItems``` method

demike, 2020-06-03:
https://github.com/node-opcua/node-opcua/blob/0d168bdc8351bbaf7ff9e60e21d459e8c4bdfeca/packages/node-opcua-client/source/client_session.ts#L412

mikakaraila, 2020-06-03:
You can use monitedItem.terminate(); not obvious but working.

demike, 2020-06-03:
> You can use monitedItem.terminate(); not obvious but working.

Yes but that's no batch request

erossignon, 2020-06-03:
@demike, well spoted !   deleteMonitoredItems is not exposed in the typescript interface.

However,
- the method exists and can be called if needed. 
- as @mikakaraila  pointed out, deleteMonitoredItems is proably not needed in most cases as the subscription will be handled with the ClientSubscriputon object that will take care of dealing with the monitoredItem  for you .

demike, 2020-06-03:
Yes of course. Just spotted it when writing some e2e tests not using the createSubscription2 approach.

demike, 2020-06-05:
Should be fixed now

closed 2020-06-05 by demike

erossignon, 2020-06-05:
Not released yet ... but not far !