node-opcua/node-opcua #802: Have null check for all variables

aki-0919, 2020-06-08:
Hi,

For safety use, it's better to have null check for all variables which are used in this function.

```
if (!_sessionPriv.$clientAlarmList) {
    return;
}
```
↓
```
if (!_sessionPriv.$clientAlarmList
     || !_sessionPriv.$monitoredItemForAlarmList
     || !_sessionPriv.$subscriptionforAlarmList ) {
    return;
}
```

https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/source/alarms_and_conditions/client_alarm_tools.ts#L45

erossignon, 2020-06-12:
feel free to provide a PR with the suggested change

iliareshetov, 2023-04-12:
Hi @erossignon 

After carefully reviewing the code, it seems that the issue you've raised is actually irrelevant. In the given context, there is no scenario where an undefined reference exception could be triggered. The current check if (!_sessionPriv.$clientAlarmList) { return; } is more than sufficient to ensure the safety of the function execution.

As a result, there is no need to add additional null checks for the other variables, and we can consider this task closed.

closed 2023-04-12 by erossignon