node-opcua/node-opcua #803: Let Crawler assign raw object of DataValue & DataType instead of convert to string

bruegth, 2020-06-09:
**Is your feature request related to a problem? Please describe.**
Reuse of crawled dataValue and datatype objects is not possible because they will converted to string.

**Describe the solution you'd like**
Assign DataValue & DataType as it is. 

**Describe alternatives you've considered**
Implement an option in crawler to make String conversion.

**Additional context**
https://github.com/node-opcua/node-opcua/blob/bca147370c37c5615678e23dca514bef0f63c15c/packages/node-opcua-client-crawler/source/node_crawler.ts#L879

erossignon, 2020-06-13:
fixed in node-opcua@2.5.6 Thomas, can you confirm it's working as expected.
( note; dataTypes remain as string and can be easilu converted back to NodeId using resolveNodeId)

bruegth, 2020-06-15:
> fixed in node-opcua@2.5.6 Thomas, can you confirm it's working as expected ?
> ( note; dataTypes remain as string and can be easilu converted back to NodeId using resolveNodeId)

ToJSON will convert enum's to string and not to related numbers.

So instead of using resolveNodeId, I do this:
OpcUa.DataType[node.dataValue.value.dataType as keyof typeof OpcUa.DataType]

Because this can be also applied to VariantArrayType,  because it's also string and can't be converted by resolveNodeId.

erossignon, 2020-06-15:
I mentioned resolveNodeId to rebuild a nodeId from the dataType property of the UAVariable, that you were referring node-opcua/packages/node-opcua-client-crawler/source/node_crawler.ts Line 879 in bca1473
not he dataType property of the Variant object.

bruegth, 2020-06-17:
That's correct I refer only DataType but I need both DataType and DataValue. DataValue.ToJSON() will convert underlying  DataValue.Value.dataType to e.g. "Double" so but for both cases "x as keyof typeof DataType" will convert it back to Datatype in every case.

So I can work with your change, and no longer need a custom version of crawler.

erossignon, 2020-06-17:
Thank you Thomas. Using enumeration value as string was a design decision to make it more user friendly as number can be confusing. I am glad that we have a easy way to convert enum values string<->value with Typescript.

closed 2020-06-17 by erossignon