node-opcua/node-opcua #811: error while doing npm install

bblumberg, 2020-06-15:
**Current behavior**

After the 2.6.6 update, if you try doing npm install node-opcua you get a fatal error

**Describe the bug**
```
 npm install node-opcua
npm ERR! Unexpected end of JSON input while parsing near '...service-endpoints":"^'

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/brbl/.npm/_logs/2020-06-15T12_35_47_347Z-debug.log
```
Log file is attached. 

**To Reproduce**
Steps to reproduce the behavior:
1. ....npm install node-opcua
2. ......
3.......

**Expected behavior**
A clear and concise description of what you expected to happen.
Installs without errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Gist**
gist:

 * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * ( )  my request is related to node-opcua acting as a OPCUA SERVER 

  * ( ) I have installed node-opcua from source ( using git clone)
  * ( x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - (x ) MacOs   : version : ____10.14.6_____
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     

* **node-opcua version:**  :

2.6.6

* **Node:** 
   `node --version` = 12.9.0
npm --version = 6.10.2
[log.log](https://github.com/node-opcua/node-opcua/files/4780271/log.log)

erossignon, 2020-06-15:
I cannot reproduce here from fresh installed on MacOs:
can you try this again and confirm that you can consistently reproduce the issue ?

```
$ mkdir tmpFolder
$ cd tmpFodler
$ npm init
$ npm install node-opcua@2.6.6 
```

On my side, the process performs well, I wonder if npm repos was stable at the time you fecth the data ?

May be worth deleted your node_modules folders and package-lock.json and redo npm install .

bblumberg, 2020-06-16:
Hi Etienne

Hmmm... I saw the same behavior when I replicated the steps above. It's
very odd isn't it?

Thanks for all of your work on node opcua (and especially the book)

erossignon, 2020-06-16:
```npm cache clean --force``` might help

bblumberg, 2020-06-16:
Thanks. That fixed the problem. I will close the issue on GitHub.

bb

closed 2020-06-16 by bblumberg