node-opcua/node-opcua #812: 2.6.6 breaks with Siemens S7-1500 PLC, 2.6.4 works

teropes, 2020-06-15:
Greetings,

I have been running with v2.6.4 for a while now and it has worked against Siemens S7-1500 PLC OPC UA server. I just did fresh npm install where I point to exact version "node-opcua": "2.6.4", but running with that version breaks. I checked package-lock file and it seems there are lots of references to 2.6.6 in my 2.6.4 installation. My previous project structure has working package-lock with 2.6.4 versions.

The error I'm getting looks like this. I ran my project with DEBUG=client-dynamic-extension-object and when I'm trying to read one nodeId, I get only empty object as read result and this error in DEBUG output:
```
15:28:57.689Z :client_dynamic_extension_objec:397   in ... populateDataTypeManager
15:28:57.702Z :client_dynamic_extension_objec:411   namespaceArray   0  :http://opcfoundation.org/UA/  1  :urn:SIMATIC.S7-1500.OPC-UAServer:181V09-A10  2  :http://opcfoundation.org/UA/DI/  3  :http://www.siemens.com/simatic-s7-opcua
15:28:57.715Z :client_dynamic_extension_objec:442   Good (0x00000)
15:28:57.716Z :client_dynamic_extension_objec:443   Opc.Ua 2:Opc.Ua.Di 3:SimaticStructures
15:28:57.716Z :client_dynamic_extension_objec:450   found 2 dictionary
15:28:57.724Z :client_dynamic_extension_objec:40    Cannot find Deprecated property for dataTypeDictionary ns=2;i=6435
{}
```

Pieces of package-locks to determine versions:
**works**
```
    "node-opcua-address-space": {
      "version": "2.6.4",
      "resolved": "https://registry.npmjs.org/node-opcua-address-space/-/node-opcua-address-space-2.6.4.tgz",
      "integrity": "sha512-3kzyOKstvVG/hjcT+S/wZmkjbwqGbjrcCQY7jbK7XrINcAifKWNRvxfPRsp3mWfkU7UzZO5PmfEUwu0M3fstQg==",
      "requires": {
        "async": "^3.2.0",
        "chalk": "^4.0.0",
        "dequeue": "^1.0.5",
        "node-opcua-assert": "^2.6.1",
        "node-opcua-basic-types": "^2.6.1",
        "node-opcua-client-dynamic-extension-object": "^2.6.4",
        "node-opcua-common": "^2.6.4",
        "node-opcua-constants": "^2.6.1",
        "node-opcua-crypto": "^1.3.2",
        "node-opcua-data-access": "^2.6.4",
        "node-opcua-data-model": "^2.6.1",
        "node-opcua-data-value": "^2.6.4",
        "node-opcua-date-time": "^2.6.1",
        "node-opcua-debug": "^2.6.1",
        "node-opcua-enum": "^2.6.1",
        "node-opcua-factory": "^2.6.1",
        "node-opcua-nodeid": "^2.6.1",
        "node-opcua-numeric-range": "^2.6.1",
        "node-opcua-object-registry": "^2.6.1",
        "node-opcua-pseudo-session": "^2.6.4",
        "node-opcua-schemas": "^2.6.4",
        "node-opcua-service-browse": "^2.6.4",
        "node-opcua-service-call": "^2.6.4",
        "node-opcua-service-filter": "^2.6.4",
        "node-opcua-service-history": "^2.6.4",
        "node-opcua-service-translate-browse-path": "^2.6.4",
        "node-opcua-service-write": "^2.6.4",
        "node-opcua-status-code": "^2.6.1",
        "node-opcua-types": "^2.6.4",
        "node-opcua-utils": "^2.6.1",
        "node-opcua-variant": "^2.6.1",
        "node-opcua-xml2json": "^2.6.4",
        "pretty-error": "^2.1.1",
        "set-prototype-of": "^1.0.0",
        "thenify": "^3.3.0",
        "underscore": "^1.10.2",
        "xml-writer": "^1.7.0"
      }
    },
```

**is broken**
```
 "node-opcua-address-space": {
      "version": "2.6.6",
      "resolved": "https://registry.npmjs.org/node-opcua-address-space/-/node-opcua-address-space-2.6.6.tgz",
      "integrity": "sha512-px4GFpwNwgpYf8bEeNQrae0j3AWCz4nWL+4fj2qKNbaGqS1oO19wRs87v19RKv+QmBTjToeo3T15jkYEDONkBg==",
      "requires": {
        "async": "^3.2.0",
        "chalk": "^4.0.0",
        "dequeue": "^1.0.5",
        "node-opcua-assert": "^2.6.1",
        "node-opcua-basic-types": "^2.6.1",
        "node-opcua-client-dynamic-extension-object": "^2.6.5",
        "node-opcua-common": "^2.6.5",
        "node-opcua-constants": "^2.6.1",
        "node-opcua-crypto": "^1.3.2",
        "node-opcua-data-access": "^2.6.5",
        "node-opcua-data-model": "^2.6.5",
        "node-opcua-data-value": "^2.6.5",
        "node-opcua-date-time": "^2.6.1",
        "node-opcua-debug": "^2.6.1",
        "node-opcua-enum": "^2.6.1",
        "node-opcua-factory": "^2.6.5",
        "node-opcua-nodeid": "^2.6.1",
        "node-opcua-numeric-range": "^2.6.5",
        "node-opcua-object-registry": "^2.6.1",
        "node-opcua-pseudo-session": "^2.6.5",
        "node-opcua-schemas": "^2.6.5",
        "node-opcua-service-browse": "^2.6.5",
        "node-opcua-service-call": "^2.6.5",
        "node-opcua-service-filter": "^2.6.5",
        "node-opcua-service-history": "^2.6.5",
        "node-opcua-service-translate-browse-path": "^2.6.5",
        "node-opcua-service-write": "^2.6.5",
        "node-opcua-status-code": "^2.6.1",
        "node-opcua-types": "^2.6.5",
        "node-opcua-utils": "^2.6.1",
        "node-opcua-variant": "^2.6.5",
        "node-opcua-xml2json": "^2.6.4",
        "pretty-error": "^2.1.1",
        "set-prototype-of": "^1.0.0",
        "thenify": "^3.3.0",
        "underscore": "^1.10.2",
        "xml-writer": "^1.7.0"
      }
    },
```

It is a bit worrying that when project relies on the main packages fixed version such as 2.6.4 without any ^ carets, still the underlying sub package versions get bumped and might break.

erossignon, 2020-06-15:
This shouldn't have been a breaking change..  
"node-opcua-client-dynamic-extension-object": "^2.6.5" has been adjusted to work with "Beckhoff Plc" extension objects but introduced a regression.  version 2.6.7 will fix this.
Thank you for testing the siemens case for us, well appreciated !

teropes, 2020-06-18:
I can confirm that 2.6.7 fixed the regression. Thanks for quick patch!

closed 2020-06-18 by teropes