node-opcua/node-opcua #819: monitoredItem.on("changed", (dataValue) => {...});

korpmaks, 2020-06-27:
how can I get monitoredItemId in '(dataValue) =>' function.
this doesnt work.

teropes, 2020-06-27:
"changed" event provides the dataValue. I think it's assumed that where you register the listener you also have the nodeId available and you can pass it to your callback if it is needed. I have little wrapper function around subscription, here is how I use it:

```
  async subscribe(nodeId, dataChanged, options, indexRange) {
    const self = this;

    // fail if not ready
    if (!self.dataSubscription || !self.opcSession) return Error("data subscription or session not ready for subscription.");

    const defaultOptions = {
      samplingInterval: 100,
      publishingInterval: 200,
      discardOldest: true,
      queueSize: 1
    };
    
    // use defaults and assign passed monitoring options on top of it
    let monitoredItemOptions = options || {};
    _.defaults(monitoredItemOptions, defaultOptions);

    let range = indexRange ? indexRange : null;
    
    // support passing nodeId without ns=, default to "self.ns" which is my opc ua client class property = our app default ns
    const completeNodeId = (!nodeId.includes("ns=")) ? `ns=${self.ns};${nodeId}` : nodeId ;

    const itemToMonitor = {
      nodeId: completeNodeId,
      attributeId: opcua.AttributeIds.Value,
      indexRange: range
    };

    const monitoredItem = await self.dataSubscription.monitor(itemToMonitor, monitoredItemOptions, opcua.TimestampsToReturn.Server);

    // bind provided dataChanged to data changes
    monitoredItem.on("changed", async function(dataValue) {
      // note, i pass nodeId to my callback, it allows using same callback for multiple different nodeId's and it can be switch cased  in the callback to differentiate between nodes
      await dataChanged(dataValue.value.toJSON().value, nodeId);
    });

  }
```

call from application:
```
// register listener
app.connectors.opcua.subscribe(AC.OPCUA.NODE_SAFETY, _.bind(app.safetyChanged, app), subscriptionOptions);

// callback gets data as plain javascript object and nodeId
async safetyChanged(safety, nodeId) {
  console.log(safety); 
  console.log(nodeId); 
});
```

example of registering to multiple nodeIds using same callback and handling them differently:
```
// register listener
app.connectors.opcua.subscribe(AC.OPCUA.INPUT_SENSOR, _.bind(app.sensorChanged, app), subscriptionOptions);
app.connectors.opcua.subscribe(AC.OPCUA.OUTPUT_SENSOR, _.bind(app.sensorChanged, app), subscriptionOptions);

// callback gets data as plain javascript object and nodeId
async sensorChanged(sensor, nodeId) {
  switch(nodeId) {
    case AC.OPCUA.INPUT_SENSOR:
      console.log("input sensor changed");
      console.log(sensor);
      break;
    case AC.OPCUA.OUTPUT_SENSOR;
      console.log("output sensor changed");
      console.log(sensor);
      break;
  }
});

```

Hope it helps.

korpmaks, 2020-07-02:
// it helps, 'let' seems to remember current instance of monitoredItem
for (let i in conf.opcua.nodes) {
   ...
   let monitoredItem = await opcuaSubscription.monitor(itemToMonitor, conf.opcua.monitoredParameters, 
   opcua.TimestampsToReturn.Both);
   monitoredItem.on("changed", (dataValue) => {
      let nodeId=conf.opcua.nodesPrefix+monitoredItem.itemToMonitor.nodeId.value;
      ...
   };
};

korpmaks, 2020-07-02:
//mqtt library () is more convinient for batch processing, can you do the same? Are there any exapmles for 'publish' ?
`mqttClient.subscribe(topicArr, function (e) {
						if (e) console.error('mqttClient.subscribe'+e.stack)
						else mqttClient.on('message', function (topic, message) {
							switch (topic) {
                                                          ...
                                                        };
						});
					});`

erossignon, 2020-07-16:
In short, you can use a javascript closure pattern to capture the monitoredItem object and have it available inside your event handler lambda function. 
````
for (let i in conf.opcua.nodes) {
   // having the  monitoredItem define in inner scope of the for loop, create a closure.
   let monitoredItem = await opcuaSubscription.monitor(itemToMonitor, conf.opcua.monitoredParameters, 
   opcua.TimestampsToReturn.Both);
   monitoredItem.on("changed", (dataValue) => {
      let nodeId=conf.opcua.nodesPrefix+monitoredItem.itemToMonitor.nodeId.value;
      ...
   };
};
```

The key aspect here is to understand how ["Javascript Closure"](https://www.tutorialsteacher.com/javascript/closure-in-javascript) works.

```javascript
function test() {

    let a;
    for (let i of [1, 2, 3, 4, 5]) {
        a = { i: i };
        setImmediate(() => {
            console.log(a);
        });
    }
    /* will produce
    { i: 5 }
    { i: 5 }
    { i: 5 }
    { i: 5 }
    { i: 5 }
    */
    for (let i of [1, 2, 3, 4, 5]) {
        let a = { i: i }; // a is inside a closure
        setImmediate(() => {
            console.log(a);
        });
    }
    /* will produce
    { i: 1 }
    { i: 2 }
    { i: 3 }
    { i: 4 }
    { i: 5 }
    */

}
````

Alternatively. you can also access the monitored item by using a function instead of a lambda.
In this case the this object will be the corresponding monitoredItem
```javascript
  monitoredItem.on('changed', function(value) {
      // note, in the context of this function , the this object correspond to the monitoredItem you need
     console.log("Id", this.monitoredItemId, ' New Value = ', value.toString());
   });
```

closed 2020-11-30 by erossignon